ipmitool/ipmitool #232: Failed to initialize the OEM info dictionary

roadglide03, 2020-09-04:
ipmitool was working great, but recently when I try to launch I get this:  Failed to initialize the OEM info dictionary

I have recompiled but still same issue.  The company recently made us upgrade Windows 10 to the latest release due to support phasing out for the earlier release.  This is the only change that I know of.  Let me know if you have anything I can try.  BTW it does compile so using the new ipmitool,exe release I made sure of that.  thanks Andy

 /usr/local/bin/ipmitool.exe  -U root -H 10.0.1.199 -P PASSWORD chassis power status
Failed to initialize the OEM info dictionary

AlexanderAmelkin, 2020-09-04:
Please see #135. You don't `enterprise-numbers` file installed. Usually `sudo make install` does that for you.

roadglide03, 2020-09-04:
I did that but same thing, I will try again and post info from that.

roadglide03, 2020-09-04:
May be helpful forgot ti mention this is in Cygwin, thanks

roadglide03, 2020-09-04:
Same issue and its installing the enterprise-numbers.  I am not a developer by trade so maybe I am missing something, but didn't have this problem until recently was working fine.:

Connecting to cso.att.com (cso.att.com)|135.188.17.109|:80... connected.
HTTP request sent, awaiting response... 200 OK
Length: 13426 (13K) [text/html]
Saving to: ‘STDOUT’

-                                      100%[===========================================================================>]  13.11K  --.-KB/s    in 0.1s

2020-09-04 11:13:03 (131 KB/s) - written to stdout [13426/13426]

mkdir -m 755 -p /usr/local/share/misc
/usr/bin/install -c -m 644 enterprise-numbers /usr/local/share/misc/
mkdir -p /usr/local/share/doc/ipmitool
/usr/bin/install -c -m 644 ./README ./COPYING ./AUTHORS ./ChangeLog /usr/local/share/doc/ipmitool
rm enterprise-numbers
make[2]: Leaving directory '/cygdrive/c/Users/ad752h/Documents/GitHub/ipmitool'
make[1]: Leaving directory '/cygdrive/c/Users/ad752h/Documents/GitHub/ipmitool'

roadglide03, 2020-09-14:
file is there BTW this is Cygwin: 

[ andy /usr/local/share/misc]: ls -l
total 16
-rw-r--r-- 1 andyDomain Users 13426 Sep  4 11:13 enterprise-numbers

roadglide03, 2020-09-16:
> Please see #135. You don't have the `enterprise-numbers` file installed. Usually `sudo make install` does that for you.

It builds just tried again but still doesn't work in Cygwin.  Something changed becuase it used to.  The only thing that was updated was Windows 10 to Version 1809

====
Saving to: ‘STDOUT’

-                   100%[===================>]  13.11K  --.-KB/s    in 0.1s

2020-09-16 09:42:17 (119 KB/s) - written to stdout [13426/13426]

mkdir -m 755 -p /usr/local/share/misc
/usr/bin/install -c -m 644 enterprise-numbers /usr/local/share/misc/
mkdir -p /usr/local/share/doc/ipmitool
/usr/bin/install -c -m 644 ./README ./COPYING ./AUTHORS ./ChangeLog /usr/local/share/doc/ipmitool
rm enterprise-numbers
make[2]: Leaving directory '/cygdrive/c/Users/ad752h/Documents/GitHub/ipmitool'
make[1]: Leaving directory '/cygdrive/c/Users/ad752h/Documents/GitHub/ipmitool'

[GACDTL12BB9263ad752h ~/Dist/Dev/GitHub/ipmitool]: ls -l /usr/local/bin
total 3420
-rwxr-xr-x 1 ad752h Domain Users 3500847 Sep 16 09:42 ipmitool.exe*
[GACDTL12BB9263ad752h ~/Dist/Dev/GitHub/ipmitool]: /usr/local/bin/ipmitool.exe
Failed to initialize the OEM info dictionary

roadglide03, 2020-09-16:
Wondering if its because this is a work computer that something was changed on the proxy side or security side that is preventing this from working?  Would anything like that cause the error that I am seeing?

AlexanderAmelkin, 2020-09-16:
It definitely has nothing to do with proxy or any other network settings. It needs a local file called `/usr/local/share/misc/enterprise-numbers`. Check that it exists in your Cygwin environment. I'm not very familiar with Windows or Cygwin.

roadglide03, 2020-09-17:
> It definitely has nothing to do with proxy or any other network settings. It needs a local file called `/usr/local/share/misc/enterprise-numbers`. Check that it exists in your Cygwin environment. I'm not very familiar with Windows or Cygwin.

It does I noted that in one of the messages and its populated.

AlexanderAmelkin, 2020-09-17:
probably, the `ipmitool` binary on Windows is looking for it in some other location. Maybe it has to use the native windows path to the file? Can you try to configure ipmitool with `./configure IANADIR=C:/Temp/` or something like that?

AlexanderAmelkin, 2020-09-22:
@roadglide03, ping. Has the problem got resolved for you? Please share the details if so.

shuang4intel, 2022-01-16:
I observed same issue "Failed to initialize the OEM info dictionary"
I followed suggestion did make install and ./configure IANADIR=C:/Temp/  it fixed the issue.  Thanks !!
Previously, IANADIR = ./

closed 2022-01-25 by AlexanderAmelkin

dwillis77, 2022-11-11:
An update for v1.8.19 - I was getting this same error when running ipmitool on CYGWIN, and after some debugging I found that the IANA_PEN URL is wrong in `Makefile.am` (it looks like it was downloading the HTML for the "GUI version" of the PEN page to enterprise-numbers instead of the raw text version). Correcting it to `IANA_PEN = https://www.iana.org/assignments/enterprise-numbers.txt` prior to running `./bootstrap`, `./configure`, `make` and `make install` resolved the issue for me (it appears `https://www.iana.org/assignments/enterprise-numbers/enterprise-numbers` would work as well).

I did not have to make any change to IANADIR in my case (with the default config it was able to read `enterprise-numbers` from `/usr/local/share/misc`) although it wouldn't hurt to do so.

Math8323, 2022-11-29:
On mac OS:
> ipmitool 
> Failed to initialize the OEM info dictionary
> brew uninstall ipmitool
> git clone https://github.com/ipmitool/ipmitool.git
> cd ipmitool
> ./bootstrap && ./configure && make && sudo make install
Now it's ok for me

AlexanderAmelkin, 2022-11-30:
@dwillis77, that problem has been fixed on master branch recently.

AlexanderAmelkin, 2022-11-30:
@Math8323, it looks like the brew package lacks the IANA database copy or installs it into some wrong place. It may as well be related to the problem with IANA's changed URL (fixed on master)

SquidlyMan, 2022-12-13:
I see this issue on macOS Ventura (on M1).  I don't have an issue with 1.8.18_3 on macOS Ventura on Intel.  I know there's a pathing change with HomeBrew on M1, so I'm trying to troubleshoot a bit more.

gylove1994, 2022-12-17:
> On mac OS 13+:
> 
> > ipmitool
> > Failed to initialize the OEM info dictionary
> > brew uninstall ipmitool
> > git clone https://github.com/ipmitool/ipmitool.git
> > cd ipmitool
> > ./bootstrap && ./configure && make && sudo make install
> > Now it's ok for me

Sorry it seems not work for me(m1 max),some error logs here:
```text
/bin/sh ../libtool --silent  --tag=CC   --mode=link gcc  -g -O2 -Wall -Wextra -std=gnu11 -pedantic -Wformat -Wformat-nonliteral   -o ipmitool ipmitool.o ipmishell.o ../lib/libipmitool.la plugins/libintf.la
ld: warning: pointer not aligned at address 0x1000A5FD3 ('_vFlash_completion_code_vals' + 19 from ../lib/.libs/libipmitool.a(ipmi_delloem.o))
ld: warning: pointer not aligned at address 0x1000A5FCA ('_vFlash_completion_code_vals' + 10 from ../lib/.libs/libipmitool.a(ipmi_delloem.o))
ld: warning: pointer not aligned at address 0x1000A5FC1 ('_vFlash_completion_code_vals' + 1 from ../lib/.libs/libipmitool.a(ipmi_delloem.o))
ld: unaligned pointer(s) for architecture arm64
clang: error: linker command failed with exit code 1 (use -v to see invocation)
make[3]: *** [ipmitool] Error 1
make[2]: *** [all-recursive] Error 1
make[1]: *** [all-recursive] Error 1
make: *** [all] Error 2
```