ipmitool/ipmitool #233: EXEC change  the LAN ip settings from remote

wangcrazy1, 2020-09-09:
I want to use the exec option to change the ip from remote. I have found if the new ip can't ping to the source ip address even using the exec option the command can't run all it.
example:
# cat lansetup.scr
lan set 7 ipsrc static
lan set 7 ipaddr 10.1.1.10
lan set 7 netmask 255.255.255.0
lan set 7 defgw ipaddr 10.1.1.254
#local ip is 192.168.1.2
# ipmitool -I lanplus -H 192.168.1.1 -U admin -P password exec lansetup.scr
only the ip will change. netmask and default gateway will not change. Is there any method can do it?

AlexanderAmelkin, 2020-09-09:
@wangcrazy1 What happens if you execute those commands one by one manually, not using exec ? Do they work that way with your BMC ?

wangcrazy1, 2020-09-11:
> @wangcrazy1 What happens if you execute those commands one by one manually, not using exec ? Do they work that way with your BMC ?

If I manually input it the ip will change after long time no response. Then I need change my local ip for connect to new ILO ip do the remain operation. I suppose the exec should be a transaction but look like it didn't work. Is there any way can send the serial command to remote ILO ip to run not just interactive every command？