ipmitool/ipmitool #238: ipmitool not working on centos8 but on 7

soofff, 2020-09-18:
hi, i am using ipmitool with docker centos 8 and 7.

`ipmitool -v -L user  -I lanplus -H my.ipmi.host p -U my_ipmi_user chassis status` with centos 8:

```
Using best available cipher suite 3

Error sending request.
Get Device ID command failed
Running Get PICMG Properties my_addr 0x20, transit 0, target 0x20
No response from Get PICMG Properties
Running Get VSO Capabilities my_addr 0x20, transit 0, target 0x20
No valid response received
Discovered IPMB address 0x0
Error sending Chassis Status command
Close Session command failed

``` 

With centos7 i am getting the correct response. This is maybe an os related issue but does somebody know what could be the reason for it?

`ipmitool version 1.8.18` on 7 and 8

thanks and br

AlexanderAmelkin, 2020-09-18:
Please use the latest code from this repository, not the version that is provided in CetnOS repos.

Silentman7979, 2020-12-19:
Hi AlexanderAmelkin,

Where could I get the latest version? This project has daily build or weekly build or a build for the new commit, doesn't it? And if it has these builds, I could get the link to download this build!

Thanks,

doublechiang, 2020-12-19:
Have you tried build by yourself?

Silentman7979, 2020-12-20:
Yes, I built ipmitool completely. Sometime, I want download it (the latest build under master branch) for my demo or simulation. So it will be good if I download directly from the internet. Because, when I download the exist build from the internet, I could use it immediately, and save time to install necessary packages for compiling this tool. This is just my opinion!

AlexanderAmelkin, 2021-02-09:
@Silentman7979, this project does have builds for commits, but the builds do not produce public artifacts, they are for verification only.

If you're willing to set up packaging of builds for different operating systems and their versions, you're most welcome to do so using GitHub Actions and publish a pull request. I would gladly merge it and have binaries published for each commit into the master branch.

Silentman7979, 2021-02-10:
@AlexanderAmelkin , I will think about your suggestion. Thanks for your feedback.