ipmitool/ipmitool #242: Unable to get ifo about single DIMM

alexyack, 2020-09-30:
I am using `ipmitool` version 1.8.18-8 on Ubuntu 20.04

When issuing command to list all FRUs - info is read successfully:

```
$ ipmitool -I lanplus -H x.x.x.x -U *** -P *** -v fru

FRU Device Description : CPU 1 DIMM 1 (ID 110)
 Memory Type           : DDR4 SDRAM
 SDRAM Package Type    : Monolithic DRAM Device
 Technology            : RDIMM
 SDRAM Die Count       : 1
 SDRAM Capacity        : 4096 Mb
 Memory Bank Group     : 2 (4 Bank Groups)
 Memory Banks          : 2 (4 Banks)
 Primary Bus Width     : 64 bits
 SDRAM Device Width    : 4 bits
 Logical Rank per DIMM : 2
 Memory size           : 16384 MB
 Memory Density        : 4 Gb
 1.2 V Nominal Op      : Yes
 TBD1 V Nominal Op     : No
 TBD2 V Nominal Op     : No
 Error Detect/Cor      : 8 bits
 Manufacturer          : Hewlett-Packard
 Manufacture Date      : year 2017 week  2
 Serial Number         : 1566035e
 Part Number           : 752369-081          

SPD DATA (512 bytes)
 23 11 0c 01 84 21 00 08 00 60 00 03 08 0b 80 00
 00 00 08 0d f4 03 00 00 6c 6c 6c 11 08 74 20 08
 00 05 70 03 00 78 1e 2b 2b 00 78 00 14 3c 00 00
...
```
But when I issue command to gent info only by this DIMM I get this error:
```
$ ipmitool -I lanplus -H x.x.x.x -U *** -P *** fru print 110

 Unknown FRU header version 0x23
```
Is it an error or my misprint?

AlexanderAmelkin, 2020-09-30:
Have you tried it with the latest code from this repository?

alexyack, 2020-09-30:
Yes, the same result.

AlexanderAmelkin, 2020-09-30:
Then this is probably a bug. The FRU code is unfortunately very dirty and needs heavy refactoring.