ipmitool/ipmitool #244: When will you have a new release?

clh5, 2020-10-01:
ipmitool 1.8.18 was released on SourceForge in Oct. 2016.  When do you plan to release the next version?

Thanks.

kapsh, 2020-10-08:
I second this question in the ligth of #40.

gkraft4dell, 2020-10-13:
It has been four years.   I agree there should be a new official release for the distros to reference.

- https://github.com/ipmitool/ipmitool/tags
- https://github.com/ipmitool/ipmitool/wiki/Howto-Create-Release-Tar-Bundle

samveen, 2020-10-14:
@gkraft4dell The release note attached to the Github release notes that:

> This is Release 1.8.18 freshly rebuilt from github sources.
> There is nothing new in it relative to the 1.8.18 release available on SourceForge.

The message clearly states that there aren't any changes as compared to 1.8.18 release of SF, thus the versioning is identical.

The `215 commits to master since this release` means that after this release was made, 215 more commits have been added to the sources. It states exactly what it means...

AlexanderAmelkin, 2022-08-30:
https://github.com/ipmitool/ipmitool/releases/tag/IPMITOOL_1_8_19

closed 2022-08-30 by AlexanderAmelkin

doublechiang, 2022-08-30:
Thanks Alex for the effort. ^_^.