ipmitool/ipmitool #254: sol looptest failed with "Unexpected data length received in payload activation response"

thinhp, 2020-12-03:
Hi team,

I run command `ipmitool -I lanplus -H <bmcip> -U <user> -P <password> sol looptest 1000` and receive following message:

`Error: Unexpected data length (1) received in payload activation response`

I execute this command multiple times, and receive that message on random remain loop test counter.

Adding `-vvvvv` flag to the command I see on failure, ipmitool takes a message with payload type SOL as a response for ACTIVATE PAYLOAD command

```
Decrypted this data (16 bytes)
 01 00 00 00 00 01 02 03 04 05 06 07 08 09 0a 0a
<<<<<<<<<< RECV FROM BMC <<<<<<<<<<<
< SOL sequence number     : 0x01
< SOL acked packet        : 0x00
< SOL accepted char count : 0x00
< SOL is nack             : false
< SOL xfer unavailable    : false
< SOL inactive            : false
< SOL transmit overrun    : false
< SOL break detected      : false
<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<
SOL MSG FROM BMC (4 bytes)
 01 00 00 00
Error: Unexpected data length (1) received in payload activation response
```
Per my understanding, normal response for ACTIVATE PAYLOAD command is a message with payload type IPMI. Therefore I don't know if this symptom is acceptable.

Thanks in advance,
Thinh

AlexanderAmelkin, 2020-12-03:
@thinhp Please attach (do not copy-paste) the full log output. Sol works fine usually except for crash on host reboot of OpenBMC-driven BMCs. So what you see here may well be a problem in your BMC, but to diagnose it I need the full log.

thinhp, 2020-12-04:
[ipmisollooptest.log](https://github.com/ipmitool/ipmitool/files/5641096/ipmisollooptest.log)

Hi @AlexanderAmelkin,

Here's the full log when I execute ipmitool in a script

`for ((i=0;i<100;i++));do echo "=========== loop $i ==============";ipmitool -vvvvv -I lanplus -H <bmcip> -U <user> -P <password> sol looptest 1000 2>&1; if [ $? -eq 1 ]; then break;fi;done`

Thanks,
Thinh

thinhp, 2020-12-14:
Hi @AlexanderAmelkin ,

Look like if ACTIVATE PAYLOAD comes to BMC while the host console is printing, BMC sends SOL payload message pretty close to the response of ACTIVATE PAYLOAD in term of timing, and randomly SOL message arrives at ipmitool prior to the command response, which causes this problem.

Above is my prediction after I did 2 tests:

1. Power off the chassis before starting the test, issue doesn't occur.
2. Execute a while true echo script on host console, issue occurs very quickly.

Should ipmitool check for the message payload type when expecting a response?

Thanks,
Thinh