ipmitool/ipmitool #256: ipmitool does not check FRU 1 header from fru write 1 file

bwil150n, 2020-12-07:
Expected Results: 
"Bad" FRU data should be detected prior to writing to FRU regardless if it is bad header checksum or unknown FRU header version.

Actual Results:
There is no error checking on the incoming FRU data from file.

Preparation:
1) ipmitool fru read 1 good_fru (save for later)
2) Copy good_fru to bad_fru then edit bad_fru to change values to "00000000  00 00 00 00 00 00 00 fa  c1 00 00 00 00 00 00 00"
3) md5sum bad_fru

Steps to reproduce:
1) ipmitool fru write 1 bad_fru
    a) confirm no error
2) ipmitool fru write 1 bad_fru
    a) confirm Bad header checksum error
3) ipmitool fru write 1 good_fru
    a) confirm Bad header checksum or Unknown Fru header version error
4 ipmitool fru write 1 good_fru
    a) confirm no error

Note: This focuses on FRU ID 1 but likely applies to any FRU ID.

AlexanderAmelkin, 2020-12-07:
@bwil150n I don't think it is correct to put the burden of verification of data on ipmitool during Write FRU operation. It's the BMC's area of responsibility. The BMC must respond with an error code if the supplied data is invalid, otherwise it's a security issue in the BMC.

bwil150n, 2020-12-07:
@AlexanderAmelkin, thank you for the fast reply!  Correct me if I'm wrong, but I understand the test for header checksum is imposed on data that is immediately overwritten by the new file.

AlexanderAmelkin, 2020-12-07:
@bwil150n IPMI Write FRU command does not make any difference between the header and the rest of the data.
It looks like the error you get in step 2 is the error of decoding the incoming FRU data from the remote BMC after writing that bad data there in step 1. If your BMC wasn't defective it wouldn't accept that data in the first place and you wouldn't end up with an incorrect FRU. However, as you can see in your step 3 and 4 the error message is strictly informative as it doesn't stop ipmitool from proceeding with FRU Write command.

It could be nice though to augment the error message so that it is more clear regarding what and where the error actually is, if it isn't clear yet. I don't have any devices with writable FRUs at hand, so I can't actually see what the actual error message is now.

bwil150n, 2020-12-07:
@AlexanderAmelkin, thank you again. I'm reaching out to get more information, maybe we can at least feedback some output to clarify and illustrate.

bwil150n, 2020-12-08:
Good evening @AlexanderAmelkin. 

I agree with "It could be nice though to augment the error message so that it is more clear regarding what and where the error actually is, if it isn't clear yet."

The 'state' of good_fru and 'bad_fru' files were confirmed with a visual inspection of each with hexdump -C... The presentation of "Bad header checksum" after ipmitool fru write 1 good_fru (immediately following a bad_fru write) is misleading because it is reported against the FRU image that was overwritten by 'good_fru'. For this use case, I don't care about what was on the chip, I care about the validity of the bytes that are on the chip now.

Would a terminal session with the steps reproduced help?