ipmitool/ipmitool #257: fru edit fruid field b X XXXX not consider the case of only borad domain

zengjuxiang, 2020-12-17:
In the following case, The second command( fru edit 0 field b 2 SA0531) breaks fru data, causing fru print cannot display correctly.
 
1) ipmitool -I lanplus -H 100.2.76.176 -U admin -P admin fru print 0  //in this case, fru only have Board domain.
 Board Mfg Date        : Wed Dec  2 15:20:00 2020
 Board Mfg             : Inspur
 Board Product         : Riser
 Board Serial          : 0 
 Board Part Number     : YZRI-01647-101

2) ipmitool -I lanplus -H XXXX -U xxx -P xxx fru edit 0 field b 2 SA0531
String size are not equal, resizing fru to fit new string
Read All FRU area
Fru Size       : 256 bytes
Copy to new FRU
Section Length: 40
Padding Length: 0
NumByte Change: 4
Start SecChnge: c2
End SecChnge  : 49
Start Section : 1
End Sec wo Pad: c1
End Section   : 2a
New Padding Length: -4
change_size_by_8: 1
New Padding Length: 4
change_size_by_8: 1
header.offset.board: 1
Moving Section Product, from 0 to 8
Change multi offset from 0 to 1
Moving Remaining Bytes (Multi-Rec , etc..), from 0 to 8
Updating Field : '0 ' with 'SA0531' ... (Length from '194' to '198')
Copying remaining of sections: 21 
Calculate New Checksum: ffffffca
Writing new FRU.
Done.

3) ipmitool -I lanplus -H 100.2.76.176 -U admin -P admin fru print 0  //nothing can display

4)The reason is that only board domain is not considered in frutool, and the solution can refer to the attachment code.

zengjuxiang, 2020-12-17:
[image]

zengjuxiang, 2020-12-17:
if the attachment cannot display,please refer to the code below

In the function "ipmi_fru_set_field_string_rebuild" of ipmi_fru.c

		/* Chassis type field */
		if (f_type == 'c' )
		{
			printf("Moving Section Chassis, from %i to %i\n",
						((header.offset.board) * 8),
						((header.offset.board + change_size_by_8) * 8)
					);
                                                if( 0 != header.offset.board ) //please add this condition
                                                {
			memcpy(
						(fru_data_new + ((header.offset.board + change_size_by_8) * 8)),
						(fru_data_old + (header.offset.board) * 8),
						board_len
					);
			header.offset.board   += change_size_by_8;
                                                } 
		}
		/* Board type field */
		if ((f_type == 'c' ) || (f_type == 'b' ))
		{
			printf("Moving Section Product, from %i to %i\n",
						((header.offset.product) * 8),
						((header.offset.product + change_size_by_8) * 8)
					);
                                                if( 0 != header.offset.product ) //please add this condition
                                                {
			memcpy(
						(fru_data_new + ((header.offset.product + change_size_by_8) * 8)),
						(fru_data_old + (header.offset.product) * 8),
						product_len
					);
			header.offset.product += change_size_by_8;
                                                }
		}

doublechiang, 2020-12-17:
with all due respect, the typical way is to fork your repository then create a pull request.