ipmitool/ipmitool #258: ipmitool chassis bootdev floppy options=efiboot,persistent not work

Silentman7979, 2020-12-18:


Silentman7979, 2020-12-18:
Hi everyone,

The command ipmitool chassis bootdev floppy does not work or is not supported. Anybody help me this issue. More information, I found out the information [https://www.supermicro.com/support/faqs/faq.cfm?faq=21361](url)

Thanks
mrwhite

doublechiang, 2020-12-18:
Have you tried up-to-dated master branch code?
There is a fix recently. check  #163

Silentman7979, 2020-12-19:
The ipmitool version "ipmitool version 1.8.18.211.g050f5da", which was built. So I will try against and share the result to you.

Thanks for your support.!

Silentman7979, 2020-12-22:
Thanks for your information @doublechiang !
I set the boot-option at floppy device completely. So I will close this ticket!

This is the result when I use ipmitool get the boot option

```
bootparam get 5
Boot parameter version: 1
Boot parameter 5 is valid/unlocked
Boot parameter data: a03c000000
 Boot Flags :
   - Boot Flag Valid
   - Options apply to only next boot
   - BIOS EFI boot 
   - Boot Device Selector : Force Boot from Floppy/primary removable media
   - Console Redirection control : System Default
   - BIOS verbosity : Console redirection occurs per BIOS configuration setting (default)
   - BIOS Mux Control Override : BIOS uses recommended setting of the mux at the end of POST

```

closed 2020-12-22 by Silentman7979