ipmitool/ipmitool #260: Output-print heavily affected if CPU is stressed, why?

MrM40, 2021-01-24:
If I stress the CPU and call e.g. `ipmitool sdr` the output-print is heavily affected performance-wise.
Can someone explain why?

It's not just that the system is stressed, all other programs/commands are not affected the same way, there is something special going on with IPMI.

Does ipmitool use a lot of CPU resources or is it the ipmi controller? Or are one of them just configured to execute at lowest possible CPU priority?

doublechiang, 2021-01-25:
ipmtool send the command to BMC. it's no relevance to the host's CPU performance.

MrM40, 2021-01-25:
But nonetheless when calling the command "stress" to all cores (ending in 100% usage) the ipmitool output-print is in particularly   heavily affected. Any suggestion too why that is? I assume BMC is not really using the CPU?

doublechiang, 2021-01-25:
BMC has its own CPU.

MrM40, 2021-01-25:
Please stay focused on my question, all other information/comments are rather irrelevant. 
Why is it output-print in particularly heavily affected?

doublechiang, 2021-01-26:
Can you elaborate on how 'output-print' got affected?

MrM40, 2021-01-26:
`time ipmitool sensor`

Normal
[image]

With high CPU usage:
[image]
It just take a longer time to print each letter, nothing is printed slower than other,

doublechiang, 2021-01-26:
I think I got your point now. Are you issue 'ipmitool sensor' in the system you are doing stress?
That say, you are not doing remotely 'ipmitool -H [bmc_ip] -U [user] -P [pass] sensor'?
If you use the same host, then I think it's normal since ipmitool require some computing power.

MrM40, 2021-01-26:
Yes, local system. And yes any program require some computing power. 
But only to repeat myself, it seems rather power-hungry since it is in  highly influenced, which doesn't make much sense.
I cannot reproduce this behavior in other low-cpu-power programs.

AlexanderAmelkin, 2021-03-09:
ipmitool sends a lot of requests over IPMI device to perform this "simple" task. That means that it executes a lot of syscalls, each of which is per se a point where the kernel's scheduler may switch to another process. Also, inside those syscalls there is a lot of waiting for response from the BMC, which (the waiting) is definitely the point where the control is switched to another process by the scheduler.

Any program doing I/O operations should be affected more or less similarly in these circumstances.

This is not an ipmitool bug. You can use https://github.com/ipmitool/ipmitool/discussions to discuss things like this.

closed 2021-03-09 by AlexanderAmelkin