ipmitool/ipmitool #268: ipmitool on loopback ip is not working

shitalkumar-gandhi, 2021-02-17:
Hi All,

ipmitool is working for other IPs, but for loopback ip (127.0.01) it is not working and gets failed with the following error
"Error: Unable to establish LAN session
Error: Unable to establish IPMI v1.5 / RMCP session"

/var/tmp # ./new -H 127.0.0.1 -U admin -P admin -vvv chassis power status
new version 1.8.18.212.g969ac9c

Sending IPMI/RMCP presence ping packet
send_packet (12 bytes)
 06 00 ff 06 00 00 11 be 80 00 00 00
ipmi_lan_send_cmd:opened=[1], open=[489552]
>> IPMI Request Session Header (level 0)
>>   Authtype   : NONE
>>   Sequence   : 0x00000000
>>   Session ID : 0x00000000
>> IPMI Request Message Header
>>   Rs Addr    : 20
>>   NetFn      : 06
>>   Rs LUN     : 0
>>   Rq Addr    : 81
>>   Rq Seq     : 01
>>   Rq Lun     : 0
>>   Command    : 38
send_packet (23 bytes)
 06 00 ff 07 00 00 00 00 00 00 00 00 00 09 20 18
 c8 81 04 38 0e 04 31
>> IPMI Request Session Header (level 0)
>>   Authtype   : NONE
>>   Sequence   : 0x00000000
>>   Session ID : 0x00000000
>> IPMI Request Message Header
>>   Rs Addr    : 20
>>   NetFn      : 06
>>   Rs LUN     : 0
>>   Rq Addr    : 81
>>   Rq Seq     : 01
>>   Rq Lun     : 0
>>   Command    : 38
send_packet (23 bytes)
 06 00 ff 07 00 00 00 00 00 00 00 00 00 09 20 18
 c8 81 04 38 0e 04 31
>> IPMI Request Session Header (level 0)
>>   Authtype   : NONE
>>   Sequence   : 0x00000000
>>   Session ID : 0x00000000
>> IPMI Request Message Header
>>   Rs Addr    : 20
>>   NetFn      : 06
>>   Rs LUN     : 0
>>   Rq Addr    : 81
>>   Rq Seq     : 01
>>   Rq Lun     : 0
>>   Command    : 38
send_packet (23 bytes)
 06 00 ff 07 00 00 00 00 00 00 00 00 00 09 20 18
 c8 81 04 38 0e 04 31
>> IPMI Request Session Header (level 0)
>>   Authtype   : NONE
>>   Sequence   : 0x00000000
>>   Session ID : 0x00000000
>> IPMI Request Message Header
>>   Rs Addr    : 20
>>   NetFn      : 06
>>   Rs LUN     : 0
>>   Rq Addr    : 81
>>   Rq Seq     : 01
>>   Rq Lun     : 0
>>   Command    : 38
send_packet (23 bytes)
 06 00 ff 07 00 00 00 00 00 00 00 00 00 09 20 18
 c8 81 04 38 0e 04 31
  No response from remote controller
Get Auth Capabilities command failed
Error: Unable to establish LAN session
Error: Unable to establish IPMI v1.5 / RMCP session
/var/tmp #

Can anyone help me to add loopback ip support ?

Thanks.

doublechiang, 2021-02-17:
Where is the server taking this command? This is not ipmitool related.

shitalkumar-gandhi, 2021-02-17:
> Where is the server taking this command? This is not ipmitool related.

This is ipmitool, just corrected the name.

doublechiang, 2021-02-17:
I am sorry, correct what?

shitalkumar-gandhi, 2021-02-17:
> I am sorry, correct what?

I mean corrected a name "ipmitool" earlier it was "new" by mistake.

closed 2021-02-24 by shitalkumar-gandhi