ipmitool/ipmitool #269: Build issue

seitzbg, 2021-02-18:
If I download the latest release and try to build I get:

```/bin/sh ../../../libtool --silent  --tag=CC   --mode=compile gcc -DHAVE_CONFIG_H -I. -I../../..  -I../../../include   -g -O2 -Wall -Wextra -std=gnu99 -pedantic -Wformat -Wformat-nonliteral -MT lanplus_crypt_impl.lo -MD -MP -MF .deps/lanplus_crypt_impl.Tpo -c -o lanplus_crypt_impl.lo lanplus_crypt_impl.c
lanplus_crypt_impl.c: In function ‘lanplus_encrypt_aes_cbc_128’:
lanplus_crypt_impl.c:167:17: error: storage size of ‘ctx’ isn’t known
  167 |  EVP_CIPHER_CTX ctx;
      |                 ^~~
lanplus_crypt_impl.c:167:17: warning: unused variable ‘ctx’ [-Wunused-variable]
lanplus_crypt_impl.c: In function ‘lanplus_decrypt_aes_cbc_128’:
lanplus_crypt_impl.c:242:17: error: storage size of ‘ctx’ isn’t known
  242 |  EVP_CIPHER_CTX ctx;
      |                 ^~~
lanplus_crypt_impl.c:242:17: warning: unused variable ‘ctx’ [-Wunused-variable]
make[4]: *** [Makefile:441: lanplus_crypt_impl.lo] Error 1
make[4]: Leaving directory '/home/seitz/tmp/ipmitool-1.8.18/src/plugins/lanplus'
make[3]: *** [Makefile:494: all-recursive] Error 1
make[3]: Leaving directory '/home/seitz/tmp/ipmitool-1.8.18/src/plugins'
make[2]: *** [Makefile:593: all-recursive] Error 1
make[2]: Leaving directory '/home/seitz/tmp/ipmitool-1.8.18/src'
make[1]: *** [Makefile:468: all-recursive] Error 1
make[1]: Leaving directory '/home/seitz/tmp/ipmitool-1.8.18'
make: *** [Makefile:400: all] Error 2
```

coklm, 2022-08-30:
the issue same as #161 ,This has been fixed back in 2017 by commit  [77fe563](https://github.com/ipmitool/ipmitool/commit/77fe5635037ebaf411cae46cf5045ca819b5c145)

AlexanderAmelkin, 2022-08-30:
@seitzbg, don't download "the latest release" please. Download the latest code. The latest release is outdated.

closed 2022-08-30 by AlexanderAmelkin