ipmitool/ipmitool #273: Windows Server 2019 Memory Leak

JanStaelens, 2021-03-02:
I've got my own c# application that is starting the IPMITool process with arguments to get metrics.
We are running 1_8_18 of the ipmitool.exe.

It was working fine on a windows server 2012. When upgrading to windows server 2019 I'm getting a memory leak of +- 1 GB every day. I can see the following:

Available Physical Memory: leaks with 1 Gb/day
Commit Charge Total: Leaks with 1 Gb/day
Virtual Memory: no leak.
Total Handles on the system: Leaks with 20 000/Day
Kernal Memory: leak of 100 Mb/day

Snapshot of the memory:
14 GB out of 28 GB used
14.5 GB Commited
9.7GB Cashed
540 MB Paged pool
791 MB non-Paged pool

There are no windows error events being logged.
There is no specific process that is actually leaking.

Some examples of the calls I'm doing:

 -h aa.bb.cc.dd -p 623 -u XXXXXX-p XXXXX-i lanplus sensor get "fan 4 dutycycle"
 -h aa.bb.cc.dd -p 623 -u XXXXXX-p XXXXX-i lanplus sensor get "11-PS 1 Inlet"
 -h aa.bb.cc.dd -p 623 -u XXXXXX-p XXXXX-i lanplus sdr elist
 -h aa.bb.cc.dd -p 623 -u XXXXXX-p XXXXX-i lanplus fru print

I've got a timer and will often call these queries and have 20 ipmitool processes running with different arguments.
They always all disappear again from the task manager.

Code I'm using to start an ipmitool process:
_____________________
 ProcessStartInfo details = new ProcessStartInfo(File);
        details.Arguments = Arguments;
        details.CreateNoWindow = true;
        details.UseShellExecute = false;
        details.RedirectStandardError = true;
        details.RedirectStandardInput = true;
        details.RedirectStandardOutput = true;

        process = new Process();
        process.StartInfo = details;
        if (process.Start() == false)
            Context.LogErr("Process " + Convert.ToString(Name) + " failed to start.", "Execute");

        coutReader = new OutputReader(process.StandardOutput, Context.protocol);
        cerrReader = new OutputReader(process.StandardError, Context.protocol);

        coutReader.Finished += new Event.FinishedEventDel(coutReader_Finished);
____________________

The Process was not getting Disposed at first. But after adding .Dispose the leak remained.
I've attempted to set compatibility mode on the .exe to use windows 8 but with no success.

Any ideas on what could be wrong? I'm thinking windows may have had some safety feature that cleaned up process memory that's now gone, perhaps combined with something in the ipmitool when it exits where it didn't cleanup everything?

doublechiang, 2021-03-02:
May I ask why you suspect ipmitool is the problem of memory leak? It ends when the process is terminated. OS should recycle all of the memory it allocated.

JanStaelens, 2021-03-03:
Hi,
I ended up at ipmitool because my code itself doesn't really do anything except calling the ipmitool and then displaying the data. 
We're reusing the same code and framework to poll and communication with other devices. Any other time in the past we've had issues or leaks it has been clearly visible as a leak on our specific process running and fixable on our side.

This is a strange leak, none of the processes show a leak but something is keeping the OS from clearing memory somewhere. The only difference with our other drivers is that we're using the ipmitool for communication instead of doing direct tcp communication (or http or snmp or ...) to devices. We also see the leak disappear immediately if we disable that specific driver that's calling ipmitool. It returns as soon as we turn it back on.

Kind Regards,

JanStaelens, 2021-03-05:
Hello, I invested further. With RAMMap I could see all the ipmitool.exe sticking around with 24K in page table memory but 0 memory.  It is exactly the same as this issue:
https://comp.protocols.time.ntp.narkive.com/esxEWWUV/ntpq-exe-memory-issue-with-windows-2019

In there, it mentioned someone having the same problem with IPMITool but getting a hotfix of some kind.
I'm uncertain if they're talking about something from windows specifically, or some special version of ipmitool?

doublechiang, 2021-03-06:
Looks like Windows 2019 problem?

JanStaelens, 2021-03-23:
I found several other programs that started having this issue on windows 2019. I'm also have an ongoing ticket with windows itself about it. 

I found the following conversation where a different tool (ipmiutil) was fixed, that had this issue. It mentions that windows doesn't clean up allocated memory on application exit in server 2019. So somewhere in the ipmitool.exe there is probably somewhere were code isn't explicitly calling a free().

-----
There were clues in the symptoms, that it was exactly the same amount of memory being leaked at each run. There is only one malloc() call in the ipmiutil sensor (isensor.c) code, which is for the SDR buffer cache. This was explicitly freed in all paths but one in isensor.c, and I made sure that it now explicitly calls free() for that buffer if it was allocated before exiting. Apparently previous versions of Windows would clean up allocated memory when an application exited, but not in this new version. So the other utllity must also have some memory buffer allocated which was not explicitly freed. I am surprised that Windows removed this memory protection in the new version.
-----

doublechiang, 2021-03-25:
For your reference, I ran the same command you ran on your windows server on my Linux environment with Valgrind (a memory debugging tool). And can't find the memory leak at least in the Linux env.
Do you have a similar tool to run on Windows?

[fru_print.log](https://github.com/ipmitool/ipmitool/files/6206949/fru_print.log)
[sdr_elist.log](https://github.com/ipmitool/ipmitool/files/6206950/sdr_elist.log)
[sensor_get.log](https://github.com/ipmitool/ipmitool/files/6206951/sensor_get.log)

AlexanderAmelkin, 2021-04-06:
First of all, please use the latest sources from the `master` branch here, not the 1.8.18 release, which is way too old now.
Does the issue still persist with the build made from the latest sources?

janstaelensskyline, 2022-03-09:
Forgot to come back with a result, apologies for the delay. This item can be closed. Root cause was proven unrelated to IPMITool. It can happen for every process you try to start from code.

After a deep investigation the issue was tracked directly to a problem in the windows server 2019 architecture and bad documentation for the Process class in .NET Framework.

When you write the following code to run the ipmi tool (or any executable):

		string pathTo = "C:\\IPMI\\Ipmi.exe";
		ProcessStartInfo details = new ProcessStartInfo(pathTo);
		details.Arguments = "-H 10.10.10.10 -p 623 -U userName -P password -I lanplus sensor get \"fan 4 dutycycle\"";

		details.CreateNoWindow = true;
		details.UseShellExecute = false;
		details.RedirectStandardError = true; // Triggers the leak
		details.RedirectStandardInput = true; // Triggers the leak
		details.RedirectStandardOutput = true; // Triggers the leak

			using (var process = new Process())
			{
				process.StartInfo = details;
				if (process.Start() == false) { Console.WriteLine("Failed to Start"); return;}
				Console.WriteLine("Started:" + details.Arguments);

				var coutReader = process.StandardOutput; // will leak even without this line
				var cerrReader = process.StandardError; // will leak even without this line
                         }

The above code will work fine on windows servers up to server 2019. In 2019 they changed something with "Named Pipeline"'s. 

You'll see the ipmi.exe dissappears from taskmanager, and all handles are cleaned up when the ipmi call is finished, but pointers were remaining which kept the ipmi.exe tool in shared memory unless you exit the calling application (visible in RAMMAP).

Digging deeper found:
The details.RedirectStandardError/Input/Output will create a named pipeline in the OS that redirects the output stream to the calling application.

The 'using' statement runs the .Dispose() on the Process automatically at the end. Unfortunately (and undocumented) that disposing doesn't close the error, input or output streams. I had to check the .NET sourcecode for the Process class to see they don't call the .Close() on the streams they just set them to null.

This causes pointers from the "Named Pipeline's to stay existing until the calling application itself exits.
(memory leak)

The way to fix this is to change the calling application code to 'manually' close the streams inside of the Process.

		using (var process = new Process())
			{
				process.StartInfo = details;
				if (process.Start() == false) { Console.WriteLine("Failed to Start"); return;}
				Console.WriteLine("Started:" + details.Arguments);

				var coutReader = process.StandardOutput; // will leak even without this line
				var cerrReader = process.StandardError; // will leak even without this line
                                
                                 process.StandardError.Close(); // Fixes the leak for win2019 servers and up
                                 process.StandardOutput.Close();  // Fixes the leak for win2019 servers and up
                                 process.StandardInput.Close();  // Fixes the leak for win2019 servers and up
                         }

The issue happens regardless if you use ipmitool or any other executable.

Microsoft was made aware of this, but I don't think they're intending to 'fix' anything about it.