ipmitool/ipmitool #279: Support DDR5 SPD

rhung2, 2021-03-30:
The current ipmitool cannot identify DDR5 fru. When does ipmitool support it?

AlexanderAmelkin, 2021-04-06:
@rhung2, there is definitely absolutely no relation between "DDR5 fru" and `ipmitool`. That is especially so because there is no such thing as "DDR5 fru". The DDR specifications from JEDEC for all versions of DDR say that there an SPD EEPROM on each DDRx module. SPD is absolutely NOT a FRU. SPD format is governed by JEDEC, while the FRU format is governed by Intel in their "IPMI FRU Information Storage Definition". There is no mention of any DDR in that latter specification. If you want to see DDR modules listed in the output of `ipmitool fru list`, please contact your BMC manufacturer and ask them to add decoded DDR5 SPD info (such as manufacturer name, part number, part name, and serial number) as IPMI FRU SDRs. As soon as that is done, `ipmitool` will show it to you.

AlexanderAmelkin, 2021-04-06:
For example, in YADRO [VESNIN](https://yadro.com/en/vesnin/) OpenPOWER P8 based servers, the DDR SPDs are decoded in OpenPOWER Firmware (that is like BIOS for that architecture), and then passed via IPMI system interface to the BMC, which in turn can report that information via `ipmitool fru list`.

In YADRO [VEGMAN](https://yadro.com/en/vegman/s220-s320) servers, the DDR SPDs are read by the BMC itself over PECI, presented in Redfish, but are NOT currently reported via IPMI (as of VEGMAN Firmware v1.2).

So it all depends on your system or BMC vendor and has nothing to do with `ipmitool`.

closed 2021-04-16 by AlexanderAmelkin