ipmitool/ipmitool #283: Unable to set defgw macaddr

xriri, 2021-04-07:
Hello, I am trying to setup IPMI and I believe I have everything setup except for the Default Gateway MAC. When I go to set it, it looks like the command runs ok but when I print out the lan info it still shows as 00:00:00:00:00:00.  Below I print out the config, attempt to set the default gateway MAC, and then print out the config again.  Thank you for your time and let me know if any other info would help.

```
[root@eris ~]# ipmitool lan print 1
Set in Progress         : Set Complete
Auth Type Support       : MD5
Auth Type Enable        : Callback : MD5
                        : User     : MD5
                        : Operator : MD5
                        : Admin    : MD5
                        : OEM      : MD5
IP Address Source       : Static Address
IP Address              : 10.82.113.116
Subnet Mask             : 255.255.255.0
MAC Address             : a0:42:3f:3d:96:be
SNMP Community String   : AMI
IP Header               : TTL=0x40 Flags=0x40 Precedence=0x00 TOS=0x10
BMC ARP Control         : ARP Responses Enabled, Gratuitous ARP Disabled
Gratituous ARP Intrvl   : 0.0 seconds
Default Gateway IP      : 10.82.113.1
Default Gateway MAC     : 00:00:00:00:00:00
Backup Gateway IP       : 0.0.0.0
Backup Gateway MAC      : 00:00:00:00:00:00
802.1q VLAN ID          : 1013
802.1q VLAN Priority    : 0
RMCP+ Cipher Suites     : 0,1,2,3,6,7,8,11,12,15,16,17
Cipher Suite Priv Max   : caaaaaaaaaaaXXX
                        :     X=Cipher Suite Unused
                        :     c=CALLBACK
                        :     u=USER
                        :     o=OPERATOR
                        :     a=ADMIN
                        :     O=OEM
Bad Password Threshold  : 0
Invalid password disable: no
Attempt Count Reset Int.: 0
User Lockout Interval   : 0

[root@eris ~]# ipmitool lan set 1 defgw macaddr be:f8:9f:a2:06:39
Setting LAN Default Gateway MAC to be:f8:9f:a2:06:39

[root@eris ~]# ipmitool lan print 1
Set in Progress         : Set Complete
Auth Type Support       : MD5
Auth Type Enable        : Callback : MD5
                        : User     : MD5
                        : Operator : MD5
                        : Admin    : MD5
                        : OEM      : MD5
IP Address Source       : Static Address
IP Address              : 10.82.113.116
Subnet Mask             : 255.255.255.0
MAC Address             : a0:42:3f:3d:96:be
SNMP Community String   : AMI
IP Header               : TTL=0x40 Flags=0x40 Precedence=0x00 TOS=0x10
BMC ARP Control         : ARP Responses Enabled, Gratuitous ARP Disabled
Gratituous ARP Intrvl   : 0.0 seconds
Default Gateway IP      : 10.82.113.1
Default Gateway MAC     : 00:00:00:00:00:00
Backup Gateway IP       : 0.0.0.0
Backup Gateway MAC      : 00:00:00:00:00:00
802.1q VLAN ID          : 1013
802.1q VLAN Priority    : 0
RMCP+ Cipher Suites     : 0,1,2,3,6,7,8,11,12,15,16,17
Cipher Suite Priv Max   : caaaaaaaaaaaXXX
                        :     X=Cipher Suite Unused
                        :     c=CALLBACK
                        :     u=USER
                        :     o=OPERATOR
                        :     a=ADMIN
                        :     O=OEM
Bad Password Threshold  : 0
Invalid password disable: no
Attempt Count Reset Int.: 0
User Lockout Interval   : 0

```

AlexanderAmelkin, 2021-04-07:
@xriri Besides, why do you want to set your gateway MAC? Are you sure you need it? In most cases that is not needed.

xriri, 2021-04-07:
Really I am not completely sure if I need it or not. I have needed it on my other setups that have the same motherboard and OS installed on them but I am not sure why.  Below is the output from a server that i have IPMI working on. Both servers have the same motherboard model. This seams to be the only difference. Once Server2 i can load the IPMI page in a browser and on Server1 i can not. Since i was able to set the Default Gateway MAC on server2 and not on server1 it got me thinking that was the issue. 

```
[root@Server2 ~]# ipmitool lan print 1
Set in Progress         : Set Complete
Auth Type Support       : MD5
Auth Type Enable        : Callback : MD5
                        : User     : MD5
                        : Operator : MD5
                        : Admin    : MD5
                        : OEM      : MD5
IP Address Source       : Static Address
IP Address              : 10.82.113.126
Subnet Mask             : 255.255.255.0
MAC Address             : a0:42:3f:39:d7:16
SNMP Community String   : AMI
IP Header               : TTL=0x40 Flags=0x40 Precedence=0x00 TOS=0x10
BMC ARP Control         : ARP Responses Enabled, Gratuitous ARP Disabled
Gratituous ARP Intrvl   : 0.0 seconds
Default Gateway IP      : 10.82.113.1
Default Gateway MAC     : be:f8:9f:a2:06:39
Backup Gateway IP       : 0.0.0.0
Backup Gateway MAC      : 00:00:00:00:00:00
802.1q VLAN ID          : 1013
802.1q VLAN Priority    : 0
RMCP+ Cipher Suites     : 0,1,2,3,6,7,8,11,12,15,16,17
Cipher Suite Priv Max   : caaaaaaaaaaaXXX
                        :     X=Cipher Suite Unused
                        :     c=CALLBACK
                        :     u=USER
                        :     o=OPERATOR
                        :     a=ADMIN
                        :     O=OEM
Bad Password Threshold  : 0
Invalid password disable: no
Attempt Count Reset Int.: 0
User Lockout Interval   : 0
```

xriri, 2021-04-07:
I also found the follow post on Stackexchange where they were saying that in order to connect to ipmi from outside the network the default gateway MAC was needed. Since i am coming from a 192.168.1.x network and my IPMI is set up on the 10.82.113.x network i figured this was worth a shot. 
[](https://serverfault.com/questions/424192/ipmi-cant-ping-or-remotely-connect)

I can confirm my routes are ok since i can get to the 10.82.113.126 ipmi web interface.

AlexanderAmelkin, 2021-04-08:
@xriri That advice on Stackexchange looks strange to me. The static MAC binding is only needed when your router doesn't respond to ARP who-has requests. Otherwise it is not needed.

AlexanderAmelkin, 2021-04-08:
@xriri Anyway, this has nothing to do with ipmitool. You really need to consult your system board manufacturer. Sorry.

Also, just for the sake of correct terminology, what you're referring to as "IPMI page" is probably a BMC WebUI page, because IPMI is a specific, not very human-friendly protocol that doesn't provide any "pages" and can only be spoken using ipmitool and other tools like ipmiutil.

AlexanderAmelkin, 2021-04-08:
@xriri You may want to try those other ipmi utilities to set the default gw mac, but I'm quite sure they will fail to do so as well. If, however, any of them succeed, please let me know here, that could mean a bug in ipmitool then.

xriri, 2021-04-09:
Sorry, yes, by IMPMI page I do mean the BMC WebUI page. I will see what I can do with the other tools.

PengWang-SMARTM, 2021-05-02:
@xriri I'd agree with Alexander that this is likely not an ipmitool issue.

To me, it's more likely either the BMC doesn't support the defgw macaddress to be changed using IPMI command, or that particular BMC is in a bad state that cannot process the command.

If you like to do a little more diagnostic, you can try the same command on the "good" BMC, and see what's the result. For this bad BMC, you can turn on the ipmitool debug option by adding -vvvvv to the command and check what's exactly responded by the BMC.