ipmitool/ipmitool #285: ipmitool sensor csv output?

mayur-who, 2021-04-22:
Does ipmitool sensor subcommand support csv style output? by taking a quick look at the code, it looks like it should but when I run 'ipmitool -c sensor list', it prints nothing.

I know I can get csv output from sdr but I also need threshold values which are not available in SDR.

is there a way for me to get sensor readings and thresholds in csv or any other format where I don't need to rely on a complicated/un-maintainable regex?

mayur-who, 2021-04-22:
Th esupport seems to have been added here:   https://github.com/ipmitool/ipmitool/commit/9d49a6edfe00d5f08cb2035431db23766a399527#diff-04c285fb9d598c5a29ea449338fc91f79bd6eea83e1d2f7935a3d9a6eae35b15

but it never made into 1.8.18 release?

doublechiang, 2021-04-22:
Have you tried to compile the current mainstream code?

mayur-who, 2021-04-27:
I haven't tried compiling the code in master branch, I can't use it directly in my project (it has to be released ipmitool binary)

I am sure it will work with compiling on my own. I am wondering why the changed didn't made it to 1.8.18

AlexanderAmelkin, 2021-04-27:
It didn't make it to 1.8.18 because 1.8.18 was two years earlier than that change.

mayur-who, 2021-04-27:
ohh, I was looking at the release dates on github releases page. my bad.

Is there a plan to release new version in near future? using csv output would really be beneficial to my project as I won't have to do complicated regex to parse the output.

mayur-who, 2021-04-29:
Is there a plan to release new version in near future? otherwise, I would need to modify my project to include this repo as a submodule and then compile it and use it in my monitoring tool

mayur-who, 2021-05-03:
@AlexanderAmelkin any thoughts on this? is there a plan to release a version of ipmitool soon? is tip of the master branch stable for me to use directly in my project?

AlexanderAmelkin, 2021-05-04:
@mayur-who The thing is, there are no automated tests that ensure stability of ipmitool in all its complexity. You may check open issues to see what known problems are and decide if that is suitable for you.

I do plan to release 1.8.19 soon but I can't guarantee any date as I only do this project in my spare time.

mayur-who, 2021-05-05:
Thank you @AlexanderAmelkin, due to not having automated tests, even I am worried about using tip of master branch directly in my codebase. I will do some tests and see if it is stable enough for the things that I need it to do.

closed 2021-05-05 by mayur-who