ipmitool/ipmitool #287: ipmitool lan set command incorrect wait implementation

PengWang-SMARTM, 2021-04-29:
I ran into an issue that "lan set ipaddr" command takes an unusual long time to finish with a warning saying "not match". I turned on debug and found out that's because ipmitool tries to read and verify the set result before issuing the "commit write".

I reviewed the source code, and I'd believe the current ipmitool does not expect an IPMC implementation that follows the IPMI specification like ours. The specification requires the lan parameter to be written upon the "commit write" if rollback feature is implemented. That is, the new parameter will not take effect until "commit write". Before the "commit write", the get command will always get the current setting rather than the new one from the set command. That's the problem that I'm running into.

I attached the highlighted specification and the proposed patch. The following is a brief explanation about the changes.

In the current ipmi_lanp.c implementation, the overall "set" procedure is, 
Line 438: set_lan_param():
> Line 443: obtain the lock, ipmi_lanp_lock(): Note, this implementation is contentious, I'd discuss it in a separated issue.
  >> Line 397: set in progress (i.e. lock), __set_lan_param().

> Line 444: set the lan parameter, __set_lan_param()
  >> Line 314: send the set request command, intf->sendrecv(()
  >> Line 335: or 346 **wait and check/verify the result, set_lan_param_wait()**

> Line 445: unlock, ipmi_lanp_unlock()
  >> Line 419: write commit, __set_lan_param()
  >> Line 425: write complete (i.e. unlock), __set_lan_param()

My proposed change moves the **wait and check/verify the result** step to the end following the unlock as below. Therefore, there should be no functional change effects.
set_lan_param():
> obtain the lock, ipmi_lanp_lock()
  >> set in progress (i.e. lock), __set_lan_param().

> set the lan parameter, __set_lan_param()
  >> send the set request command, intf->sendrecv(()

> unlock, ipmi_lanp_unlock()
  >> write commit, __set_lan_param()
  >> write complete (i.e. unlock), __set_lan_param()

> **wait and check/verify the result, set_lan_param_wait()**

Someone may argue the possibility of having an IPMC implementation that cannot take "commit write"/"set complete" command immediately. If we need to handle this situation, retry of "commit write"/"set complete" can be added.

Please let me know if further information / discussion is needed. Thanks!

Sorry that for some unknown reason, I cannot upload anything here. I'll try again later with a different browser.

PengWang-SMARTM, 2021-04-29:
[image]
[ipmitool-1.8.18-lib-ipmi_lanp.c.patch.txt](https://github.com/ipmitool/ipmitool/files/6402162/ipmitool-1.8.18-lib-ipmi_lanp.c.patch.txt)

Not sure why, GitHub loves Firefox more.

AlexanderAmelkin, 2021-04-30:
@PengWang-SMARTM Thanks for reporting. If you can do it, it would be better to submit a pull request instead of attaching a patch, let alone describing the change in words.

Submitting a pull request would definitely speed up the change integration process.

PengWang-SMARTM, 2021-04-30:
Never worked with GitHub before. Let me try.

PengWang-SMARTM, 2021-04-30:
Is there a quick start manual to ensure I'll follow a correct process?

AlexanderAmelkin, 2021-04-30:
@PengWang-SMARTM Please check here: https://github.com/ipmitool/ipmitool/wiki/How-to-contribute#2-contribute-the-changes

PengWang-SMARTM, 2021-04-30:
$ git push -u origin bugfix/287-fix-lan-set-wait-position
fatal: An error occurred while sending the request.
fatal: The request was aborted: Could not create SSL/TLS secure channel.
error: unable to read askpass response from 'C:/Program Files/Git/mingw64/libexec/git-core/git-gui--askpass'
Username for 'https://github.com': peng.wang@smartm.com
remote: Permission to ipmitool/ipmitool.git denied to PengWang-SMARTM.
fatal: unable to access 'https://github.com/ipmitool/ipmitool.git/': The requested URL returned error: 403

The github help is slightly different from the latest Git GUI I just installed. I'm not sure if I did thing right or not. But got the error messages above when I'm trying to push it. I'll try to figure it out another time.

By the way, the checked out code from master is slightly different from the 1.8.18 source, and it seem all Makefile.in are not there and some other configure/make files. Something not quite right. Anyway, I'll give it try again later.

AlexanderAmelkin, 2021-04-30:
@PengWang-SMARTM That's because you cloned from https instead of ssh

doublechiang, 2021-04-30:
@PengWang-SMARTM  you can't directly push to ipmitool source code. Please clone this project, make your changes, push it to your cloned code, then submit a pull request.

JohnVillalovos, 2021-04-30:
> Is there a quick start manual to ensure I'll follow a correct process?

This might help:

https://www.wikihow.com/Create-a-Pull-Request-on-Github

PengWang-SMARTM, 2021-05-01:
> 
> 
> @PengWang-SMARTM Please check here: https://github.com/ipmitool/ipmitool/wiki/How-to-contribute#2-contribute-the-changes

Thank you all for the tutoring, and I believe I went through the process, and submitted change requests.

However, this wiki how to contribute may need some updates as well.

1. I was having trouble because I didn't understand what it is and how to "Create a fork on GitHub and clone it to your machine". Some web pages just told me only how to clone, but not how to create a fork. Now, I know it's simply clicking the "fork" icon at the upper right corner on the web page. However, it's really a challenge for someone who never used git before.

2. In this instruction, before "git commit", a step called "stage" is required. The "git commit" command ended up with the following message if this step is missing.
```
Changes not staged for commit:
  (use "git add <file>..." to update what will be committed)
  (use "git restore <file>..." to discard changes in working directory)
```

3. There is no "git co" command anymore. When I didn't know what I should do, I noticed that the web page has a big green button, called "compare and pull request" or something similar. I clicked that button, and it finished everything automatically with a message "Able to merge. These branches can be automatically merged."
```
$ git co -b upstream-master upstream/master
git: 'co' is not a git command. See 'git --help'.

The most similar commands are
        commit
        clone
        log
```
4. I got a "fatal" error message with the push command. However, it's not really fatal. You mentioned that's because I did clone through https rather than ssh. However, I did not find how to do that through ssh. Those online examples I found are all using https through either CLI or GUI. Anyway, it seems that this "fatal" error is not harmful at all.

```
$ git push -u origin bugfix/287-fix-lan-set-wait-location
fatal: An error occurred while sending the request.
fatal: The request was aborted: Could not create SSL/TLS secure channel.
Enumerating objects: 7, done.
Counting objects: 100% (7/7), done.
...
```

5. "Rebase your code if necessary." Do you mean synchronize with the master branch again in case later check-in causes some conflicts? Or something else? How should I know if it's necessary? Is it only when the changes cannot be merged automatically?

6. Yesterday I used GUI, and I used CLI today. I feel GUI should be the same and slightly easier to use. I was failed yesterday because I did not do the "fork" step correctly. Therefore, I'd suggest prompting Git GUI, or at least mention it.

PengWang-SMARTM, 2021-05-01:
Since I've pushed the changes and created a pull request, I'd assume I should be able to safely remove my local branch. I tried the following, but failed. Any suggestions?
```
$ git branch -d bugfix/287-fix-lan-set-wait-location
error: Cannot delete branch 'bugfix/287-fix-lan-set-wait-location' checked out at 'C:/sandbox/ipmitool/ipmitool'
```
I plan to use the same directory for my another issue/branch. I want to delete this one to avoid any possible confusion.

I wish this could be part of your how-to wiki so new comer won't need to searching around anymore.

AlexanderAmelkin, 2021-05-01:
@PengWang-SMARTM Oh, sorry about git co. That's an alias that I use in my setup and I just put it there thoughtlessly. Will fix it and will also try to address other difficulties, although I wouldn't like to transform ipmitool contributions guide into a github or git howtos that are already plentifully available on the net.

PengWang-SMARTM, 2021-05-02:
@AlexanderAmelkin I understand your decision. However, from my point of view, the purpose of this "ipmitool contributions guide" is to help people make a contribution, not only the coding style, but also how to use the github to submit contributions.

In deed, this guide is pretty neat and easy to follow except the very first step, how to fork and then clone. I went into trouble at the beginning because Git GUI starts from clone. New dummy contributors like me would appreciate if the guide can be updated by just adding a sentence mentioning the fork icon on the web page, and adding the clone command.

Now I know how to do it right now. Therefore, it's not a problem for me to keep the guide as is. I just want to help other people who is willing to help. 

For the local cleanup after created pull request, I can simply delete the directory and create another from scratch. A little stupid and less graceful, but should work always. I don't really have  an interest to spend too much time and learn how to use the git for only submitting some patches to this project.

Maybe add a link to a good git tutoring page

JohnVillalovos, 2021-05-02:
I have to admit I agree with @AlexanderAmelkin   I don't think the purpose of the guide should be how to use GitHub. There are many documents/guides on how to use GitHub.

PengWang-SMARTM, 2021-05-02:
I feel it would not hurt anything to mention the fork icon and add a clone command to the guide under "Create a fork on GitHub and clone it to your machine". I don't think it will make the guide like a github manual.

Actually, other steps all have examples, except this very first step. It makes an old newbie like me don't know where to get start. 

However, if you disagree, it's fine to me. I just feel sad for those who are willing to contribute but scared away by learning how to use Git.

By the way, even though I know what I need to do, I still don't think I know how to use Git as what I know about ClearCase. Asking people to learn Git is totally unnecessary to make a contribution here.