ipmitool/ipmitool #302: Support parseable output, e.g. JSON

martinvonwittich, 2021-06-30:
I want to use `ipmitool` in a Perl script to identify BMCs with an unsafe default configuration (network access enabled) on customer servers. The correct way to do this seems to be to enumerate the channels (`ipmitool channel info 0..15`), identify which of them are network channels, and then to query them with `ipmitool lan print <channel>`.

Unfortunately, `ipmitool` does not seem to feature parseable output, which means I have to parse the human-readable output. Doing that properly is rather hard and prone to breakage, so a proper parseable output format like JSON would be appreciated.

AlexanderAmelkin, 2021-06-30:
`ipmitool` does not support JSON output, but it supports CSV output (the `-c` option).
Unfortunately, not with this particular command.

However, I think parsing the existing quite well structured output with `awk` is quite easy, and I think that it is no more prone to breakage than the CSV output.

Nonetheless, if you're willing to submit a pull request to add support for CSV output into `channel info` command, you're very welcome to do so.

Elbandi, 2022-10-02:
https://github.com/ipmitool/ipmitool/pull/370