ipmitool/ipmitool #303: Time for a release, maybe?

utkarsh2102, 2021-07-01:
Hello,

Thanks for your work on ipmitool so far! 🎉 

However, as I am seeing, it's been over 2 years since the last release and we've over 235 commits since then, which includes some awesome things like support for Quanta hardware (cf: PR #5), et al. How about a release for all the things that has been worked on in the last 2 years?

Thank you!

utkarsh2102, 2021-07-01:
cf: https://github.com/ipmitool/ipmitool/compare/IPMITOOL_1_8_17...master
(for the comparison)

Dutchman01, 2021-12-13:
any word on a new release pending?

Ypnose, 2022-08-26:
Do we have to wait for 3 more years to get a release ? Latest release do not include CVE-2020-5208 fix…

https://github.com/ipmitool/ipmitool/issues/162
https://github.com/ipmitool/ipmitool/issues/217
https://github.com/ipmitool/ipmitool/issues/244
https://github.com/ipmitool/ipmitool/issues/355

It was requested several times.

AlexanderAmelkin, 2022-08-30:
https://github.com/ipmitool/ipmitool/releases/tag/IPMITOOL_1_8_19

closed 2022-08-30 by AlexanderAmelkin