ipmitool/ipmitool #308: ipmitool can read DDR4 SPD, but DDR5 failed

rhung2, 2021-08-13:
# ipmitool.18 -I lanplus -H 192.168.70.212 -U USERID -P XXXXX fru print 34
 Board Mfg Date        : Mon Nov  4 20:00:00 2019
 Board Mfg             : DETA
 Board Product         : LENOVO Product
 Board Serial          : D1DG9B500NJ
 Board Part Number     : SP57A14714
 Board Extra           : XXXXXXX
 Board Extra           : 7200
 Board Extra           : 01
 Board Extra           : 8000
 Board Extra           : 0200
 Board Extra           : D1DG
 Board Extra           : 3
 Board Extra           : 4519
 Board Extra           : 7.12 (Primary version)
 Board Extra           : 7.14 (Secondary & Active version)
 Board Extra           : DPS-2400DB A
 Board Extra           : DG
 Board Extra           : 5812
 Board Extra           : 8SSP57A14714D1DG9B500NJ
 Board Extra           : 02
 Board Extra           : 5812
 Board Extra           : ffff
 Board Extra           : d012
 Board Extra           : ff00

DDR4 SPD is like this:
23 11 0C 04 85 29 91 08 00 60 00 03 18 0B 80 00
00 00 06 0D F8 3F 00 00 6E 6E 6E 11 00 6E F0 0A
20 08 00 05 00 60 18 28 28 00 78 00 14 3C 00 00
00 00 00 00 00 00 00 00 00 00 00 00 15 16 03 16
03 16 15 16 15 16 0B 16 0B 16 0B 16 03 16 00 00
00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
00 00 00 00 00 00 9C 00 00 00 00 00 E7 00 71 97
11 11 44 19 00 86 32 A0 01 75 0A 11 1D 1D 1D 1D
21 15 15 15 00 14 14 14 27 27 27 00 00 00 00 00
00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
....

But DDR5 spd is:
30 10 12 01 04 20 00 62 00 00 00 00 10 02 05 00
00 00 00 00 A0 01 F2 03 7A 0D 00 00 00 00 80 3E
80 3E 80 3E 00 7D 80 BB 30 75 27 01 A0 00 82 00
00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
10 00 80 B3 80 21 0B 2A 80 11 00 00 00 00 00 00
00 00 80 B3 C0 12 00 00 00 00 00 00 00 00 00 00
00 00 00 00 00 00 11 21 01 72 08 2A 00 00 00 00
....

Now ipmitool cannot identify DDR5 SPD.
FRU Device Description : DIMM_4 (ID 20)
 Memory Type           : Unknown (0x12)
 Memory Size    INVALID: 1, 4, 32, 0
 Voltage Intf          : 5.0V TTL
 Error Detect/Cor      : None
 Manufacturer          : Unknown (0x00)
 Serial Number         : 00000000

doublechiang, 2021-08-13:
You already posted article #279 and it has been closed.

rhung2, 2021-09-09:
double,
why was it closed?  did ipmitool support to identify DDR5 memory format?
If not, it should not be closed.

rhung2, 2021-09-09:
Also, if you need DDR5 SPEC, i could provide you.

doublechiang, 2021-09-09:
the reason is described at #279, you duplicated posted the issue.

rhung2, 2021-09-10:
@doublechiang 
https://github.com/ipmitool/ipmitool/issues/279#issuecomment-814482153
base on this comment, did ipmitool plan to support DDR5 info in fru cmd?

doublechiang, 2021-09-10:
what question you specifically have based on the comment on #279?

rhung2, 2021-09-15:
@doublechiang 
Alex said "@rhung2, there is definitely absolutely no relation between "DDR5 fru" and ipmitool. That is especially so because there is no such thing as "DDR5 fru". The DDR specifications from JEDEC for all versions of DDR say that there an SPD EEPROM on each DDRx module. SPD is absolutely NOT a FRU. SPD format is governed by JEDEC, while the FRU format is governed by Intel in their "IPMI FRU Information Storage Definition". There is no mention of any DDR in that latter specification. If you want to see DDR modules listed in the output of ipmitool fru list, please contact your BMC manufacturer and ask them to add decoded DDR5 SPD info (such as manufacturer name, part number, part name, and serial number) as IPMI FRU SDRs. As soon as that is done, ipmitool will show it to you."

it seemed that ipmitool didnt support to show DDR5 fru, right?

adx-git, 2021-09-15:
@rhung2 Your BMC needs to support DDR5 SPD, if this is not the case, there is nothing what the IPMITool can do.

rhung2, 2021-09-17:
@adx-git 
disagree, I also opened a defect on 2014-06-12 for supporting DDR4.
https://sourceforge.net/p/ipmitool/bugs/325/

But why did not you do somethings for DDR5?