ipmitool/ipmitool #324: ipmitool installation and creating a node to the BMC

TannaManohar, 2021-11-25:
Hi,
I am new to IPMI, I have installed the ipmitool in my ubuntu laptop and my BMC is MSP430F5358
Q1) How to create a node to my BMC so that I can access it through the ipmitool?
Q2) I am getting the error below I mentioned if I type any command.
            Could not open device at /dev/ipmi0 or /dev/ipmi/0 or /dev/ipmidev/0: No such file or directory
Q3) How we can implement ipmi over the SMBus?

AlexanderAmelkin, 2021-11-25:
@TannaManohar Laptops usually do not have any BMC. MSP430 is an generic MCU, not a specialized BMC chip. Are you sure that it does implement IPMI protocol?

TannaManohar, 2021-11-25:
Hi @AlexanderAmelkin 
As I need to implement it on MSP430, is there any alternate method to implement ipmi, so that my BMC can communicate with FPGA having similar communication architecture?

AlexanderAmelkin, 2021-11-25:
@TannaManohar Sure. Usually on x86 the BMC communicates over LPC or eSPI bus via KCS. However, since MSP430 doesn

AlexanderAmelkin, 2021-11-25:
You will then need to use `ipmi_ssif` driver on your host linux side and configure it properly (sorry, I can't help you with that).
When you have that done, you will have your `/dev/ipmi0` node and ipmitool will send/receive data via it.

AlexanderAmelkin, 2021-11-25:
Also, next time please open questions as 'Discussions' (https://github.com/ipmitool/ipmitool/discussions), not as issues.

closed 2021-11-25 by AlexanderAmelkin