ipmitool/ipmitool #332: make

tjbx, 2022-03-08:


AlexanderAmelkin, 2022-03-09:
Please try to use the standard build method for macOS, see [INSTALL](https://github.com/ipmitool/ipmitool/blob/master/INSTALL)

If the problem persists, please report.

Please also note that `ipmitool` was NOT designed for big-endian architectures and most probably will not work properly even if the build is successful. Make sure you're using ARM64 in LE mode.

tjbx, 2022-03-12:
I use MacPorts and it builds fine on an x86_64 system. As a work around, I've copied an x86_64 MacPorts tree that includes ipmitool to my M1 and because of Rosetta, it works fine.

AlexanderAmelkin, 2022-03-12:
Sorry, I really don't know anything about Macs, so all that is totally Greek to me. I'm glad that it works for you though.

zxombie, 2022-08-06:
The problem seems to be the `#pragma pack(1)` in `include/ipmitool/ipmi_delloem.h`. If I comment it out I can build ipmitool. I'm not sure if the pragma is needed as most of the structures in the header are already marked as packed.

tjbx, 2022-08-09:
I can verify that commenting out the pragma line allows it to compile and work in the macports environment. I haven't done extensive testing but I was able to connect to an SOL console and that's my main use.

pixelsoccupied, 2022-11-30:
@tjbx you wrote `As a work around, I've copied an x86_64 MacPorts tree that includes ipmitool to my M1 and because of Rosetta, it works fine.` ...could you share the steps on how to do that?

tjbx, 2022-11-30:
@pixelsoccupied I believe I just rsync'd /opt/local over from a same MacOS release Intel Mac and it just worked. I've since just applied the one line fix by @zxombie and am running things native now.

blacktide082, 2022-12-09:
Here's what worked for me on an M1 Max with Ventura. Install dependencies with Homebrew:

```bash
brew install automake openssl libtool freeipmi wget
```

Then clone the repo and comment out the `#pragma pack(1)`  on line 36 of `include/ipmitool/ipmi_delloem.h` like @zxombie suggested.

Afterwards, run the following:

```bash
export LDFLAGS="-L$HOMEBREW_PREFIX/opt/openssl@1.1/lib"
export CPPFLAGS="-I$HOMEBREW_PREFIX/opt/openssl@1.1/include"
autoupdate && ./bootstrap && ./configure && make && sudo make install
```

gregkonush, 2022-12-25:
@blacktide082 Thanks for your script, I confirm that this workaround works. Original error stack trace:
```bash
mv -f .deps/ipmishell.Tpo .deps/ipmishell.Po
/bin/sh ../libtool --silent  --tag=CC   --mode=link gcc  -g -O2 -Wall -Wextra -std=gnu11 -pedantic -Wformat -Wformat-nonliteral  -L/opt/homebrew/opt/openssl@1.1/lib -o ipmitool ipmitool.o ipmishell.o ../lib/libipmitool.la plugins/libintf.la -lcrypto
ld: warning: pointer not aligned at address 0x1000AE0D3 ('_vFlash_completion_code_vals' + 19 from ../lib/.libs/libipmitool.a(ipmi_delloem.o))
ld: warning: pointer not aligned at address 0x1000AE0CA ('_vFlash_completion_code_vals' + 10 from ../lib/.libs/libipmitool.a(ipmi_delloem.o))
ld: warning: pointer not aligned at address 0x1000AE0C1 ('_vFlash_completion_code_vals' + 1 from ../lib/.libs/libipmitool.a(ipmi_delloem.o))
ld: unaligned pointer(s) for architecture arm64
clang: error: linker command failed with exit code 1 (use -v to see invocation)
make[3]: *** [ipmitool] Error 1
make[2]: *** [all-recursive] Error 1
make[1]: *** [all-recursive] Error 1
make: *** [all] Error 2
```

closed 2023-01-09 by AlexanderAmelkin