ipmitool/ipmitool #335: ipmi sel delete <id>..<id> doesn't work (as expected?)

SenseiC, 2022-03-16:
[root@ml30g9d40u19r ~]# ipmitool sel info
SEL Information
Version          : 1.5 (v1.5, v2 compliant)
Entries          : 64
Free Space       : 0 bytes
Percent Used     : 100%
Last Add Time    : 01/01/1970 00:01:37
Last Del Time    : 01/01/1970 00:01:37
Overflow         : true
Supported Cmds   : 'Reserve'
[root@ml30g9d40u19r ~]# ipmitool sel list first 5
 237 | 04/17/2020 | 18:57:07 | Battery #0x1f | Failed | Asserted
 238 | 04/17/2020 | 18:59:10 | Battery #0x1f | Failed | Asserted
 239 | 08/08/2020 | 04:36:02 | Battery #0x1f | Failed | Asserted
 23a | 11/11/2020 | 20:40:29 | Battery #0x1f | Failed | Asserted
 23b | 12/13/2020 | 10:24:28 | Battery #0x1f | Failed | Asserted
[root@ml30g9d40u19r ~]# ipmitool sel delete 237
Unable to delete entry 237: Invalid command

I have tried numerous variations of ipmitool sel delete... all either return an error like the above or indicate unexpected command

doublechiang, 2022-03-16:
try 'ipmitool sel delete 0x237'

AlexanderAmelkin, 2022-03-16:
@SenseiC The Invalid command response is provided by your BMC. Most probably it doesn't support the delete command. Please consult your BMC vendor.

SenseiC, 2022-03-16:
I based my commands on the source code from the RPM (pulled from the Red Hat Enterprise Linux source RPM).  If the command isn't supported by the BMC (which I guess is my employer as I'm running this on one of our servers in our lab), then I guess I'm confused how the clear command works:

[root@ml30g9d40u19r ~]# ipmitool sel list first 5
 237 | 04/17/2020 | 18:57:07 | Battery #0x1f | Failed | Asserted
 238 | 04/17/2020 | 18:59:10 | Battery #0x1f | Failed | Asserted
 239 | 08/08/2020 | 04:36:02 | Battery #0x1f | Failed | Asserted
 23a | 11/11/2020 | 20:40:29 | Battery #0x1f | Failed | Asserted
 23b | 12/13/2020 | 10:24:28 | Battery #0x1f | Failed | Asserted
[root@ml30g9d40u19r ~]# ipmitool sel delete 237
Unable to delete entry 237: Invalid command
[root@ml30g9d40u19r ~]# ipmitool sel delete 0x237
Unable to delete entry 567: Invalid command
[root@ml30g9d40u19r ~]# ipmitool sel clear
Clearing SEL.  Please allow a few seconds to erase.
[root@ml30g9d40u19r ~]# ipmitool sel list first 5

I had STARTED opening this issue as an enhancement request (no way to "purge" older SEL entries), but then noticed the delete command.  I was trying to avoid having to report back to our customer that you have to choose between an all or nothing approach... do nothing or ipmitool sel clear.

So @AlexanderAmelkin the "Invalid command" is getting returned to the ipmitool by the underlying hardware/BMC?  Wondering how to identify what subset of commands the BMC considers "valid commands ". (somewhat thinking out loud)

SenseiC

AlexanderAmelkin, 2022-03-16:
@SenseiC 'Delete SEL Entry' and 'Clear SEL' are two different commands per IPMI 2.0 Specification.
Only the 'Clear SEL' command is specified as 'Mandatory'. The 'Delete SEL Entry' command is 'Optional'.
See 'Table 31-1 SEL Device Commands' in IPMI 2.0 Specification.

Apparently, your BMC doesn't implement the latter command.

closed 2022-03-16 by SenseiC