ipmitool/ipmitool #344: Unable to connect to VNXe3200 via IPMITool. RAKP 2 HMAC is invalid

Rennekww, 2022-06-08:
Good afternoon.
I can't connect to the SPA storage process at VNXe3200.

Using IPMITool version 1.8.18. Recently built on windows 10. I'm trying to connect from another PC, but also Windows 10.

Below is the information displayed when trying to connect.

c:\ipmitool>ipmitool.exe -vvv -I lanplus -C3 -U console -P FCNBC152200435 -H 128.221.1.252 sol activate
ipmitool version 1.8.18

Loading IANA PEN Registry...
IANA PEN registry open failed: No such file or directory

>> Sending IPMI command payload
>>    netfn   : 0x06
>>    command : 0x38
>>    data    : 0x8e 0x04

BUILDING A v1.5 COMMAND
>> IPMI Request Session Header
>>   Authtype   : NONE
>>   Sequence   : 0x00000000
>>   Session ID : 0x00000000
>> IPMI Request Message Header
>>   Rs Addr    : 20
>>   NetFn      : 06
>>   Rs LUN     : 0
>>   Rq Addr    : 81
>>   Rq Seq     : 00
>>   Rq Lun     : 0
>>   Command    : 38
<< IPMI Response Session Header
<<   Authtype                : NONE
<<   Payload type            : IPMI (0)
<<   Session ID              : 0x00000000
<<   Sequence                : 0x00000000
<<   IPMI Msg/Payload Length : 16
<< IPMI Response Message Header
<<   Rq Addr    : 81
<<   NetFn      : 07
<<   Rq LUN     : 0
<<   Rs Addr    : 20
<<   Rq Seq     : 00
<<   Rs Lun     : 0
<<   Command    : 38
<<   Compl Code : 0x00
>> SENDING AN OPEN SESSION REQUEST

<<OPEN SESSION RESPONSE
<<  Message tag                        : 0x00
<<  RMCP+ status                       : no errors
<<  Maximum privilege level            : admin
<<  Console Session ID                 : 0xa0a2a3a4
<<  BMC Session ID                     : 0x0200ca00
<<  Negotiated authenticatin algorithm : hmac_sha1
<<  Negotiated integrity algorithm     : hmac_sha1_96
<<  Negotiated encryption algorithm    : aes_cbc_128

>> Console generated random number (16 bytes)
 31 71 78 c4 cc 44 3b 45 95 bb cc 45 93 77 9a e3
>> SENDING A RAKP 1 MESSAGE

<<RAKP 2 MESSAGE
<<  Message tag                   : 0x00
<<  RMCP+ status                  : no errors
<<  Console Session ID            : 0xa0a2a3a4
<<  BMC random number             : 0x5e0c4cb451702dd2d188ddea78bdb136
<<  BMC GUID                      : 0x20000102027f00000000000000000000
<<  Key exchange auth code [sha1] : 0x167046d02953f9fe32674e602587b72afbe98970

bmc_rand (16 bytes)
 5e 0c 4c b4 51 70 2d d2 d1 88 dd ea 78 bd b1 36
>> rakp2 mac input buffer (65 bytes)
 a4 a3 a2 a0 00 ca 00 02 31 71 78 c4 cc 44 3b 45
 95 bb cc 45 93 77 9a e3 5e 0c 4c b4 51 70 2d d2
 d1 88 dd ea 78 bd b1 36 20 00 01 02 02 7f 00 00
 00 00 00 00 00 00 00 00 14 07 63 6f 6e 73 6f 6c
 65
>> rakp2 mac key (20 bytes)
 46 43 4e 42 43 31 35 32 32 30 30 34 33 35 00 00
 00 00 00 00
>> rakp2 mac as computed by the remote console (20 bytes)
 db 31 b2 22 a2 55 56 f7 d1 28 b5 9b fb d6 49 71
 27 e5 d4 0f
> RAKP 2 HMAC is invalid
Error: Unable to establish IPMI v2 / RMCP+ session

AlexanderAmelkin, 2022-06-08:
@Rennekww It looks to me like your BMC is returning a wrong RAKP2 HMAC key.

Please try an up to date source from github rather than some old 1.8.18.

Also try to experiment with different cipher suites (the -c option).

Rennekww, 2022-06-08:
I took the sources from here https://github.com/ipmitool/ipmitool and just recently, 3 days ago.
I tried connections from C0 to C17.

C0:
c:\ipmitool>ipmitool.exe -vvvvvv -I lanplus -C0 -U console -P FCNBC152200435 -H 128.221.1.252 sol activate
ipmitool version 1.8.18

Loading IANA PEN Registry...
IANA PEN registry open failed: No such file or directory
  Allocating      3 entries
  [     1] 16777214 | A Debug Assisting Company, Ltd.
  [     0]  1048575 | Unspecified

>> Sending IPMI command payload
>>    netfn   : 0x06
>>    command : 0x38
>>    data    : 0x8e 0x04

BUILDING A v1.5 COMMAND
added list entry seq=0x00 cmd=0x38
>> IPMI Request Session Header
>>   Authtype   : NONE
>>   Sequence   : 0x00000000
>>   Session ID : 0x00000000
>> IPMI Request Message Header
>>   Rs Addr    : 20
>>   NetFn      : 06
>>   Rs LUN     : 0
>>   Rq Addr    : 81
>>   Rq Seq     : 00
>>   Rq Lun     : 0
>>   Command    : 38
>> sending packet (23 bytes)
 06 00 ff 07 00 00 00 00 00 00 00 00 00 09 20 18
 c8 81 00 38 8e 04 b5
<< received packet (30 bytes)
 06 00 ff 07 00 00 00 00 00 00 00 00 00 10 81 1c
 63 20 00 38 00 01 97 04 03 00 00 00 00 09
<< IPMI Response Session Header
<<   Authtype                : NONE
<<   Payload type            : IPMI (0)
<<   Session ID              : 0x00000000
<<   Sequence                : 0x00000000
<<   IPMI Msg/Payload Length : 16
<< IPMI Response Message Header
<<   Rq Addr    : 81
<<   NetFn      : 07
<<   Rq LUN     : 0
<<   Rs Addr    : 20
<<   Rq Seq     : 00
<<   Rs Lun     : 0
<<   Command    : 38
<<   Compl Code : 0x00
IPMI Request Match found
removed list entry seq=0x00 cmd=0x38
>> SENDING AN OPEN SESSION REQUEST

>> sending packet (48 bytes)
 06 00 ff 07 06 10 00 00 00 00 00 00 00 00 20 00
 00 00 00 00 a4 a3 a2 a0 00 00 00 08 00 00 00 00
 01 00 00 08 00 00 00 00 02 00 00 08 00 00 00 00
<< received packet (52 bytes)
 06 00 ff 07 06 11 00 00 00 00 00 00 00 00 24 00
 00 00 01 00 a4 a3 a2 a0 00 cf 00 02 00 00 00 08
 00 00 00 00 01 00 00 08 00 00 00 00 02 00 00 08
 00 00 00 00
<<OPEN SESSION RESPONSE
<<  Message tag                        : 0x00
<<  RMCP+ status                       : no errors
<<  Maximum privilege level            : callback
[... 314 lines cut ...]
 d7 e1 0e 89 27 53 dd 2c
<<RAKP 2 MESSAGE
<<  Message tag                   : 0x00
<<  RMCP+ status                  : no errors
<<  Console Session ID            : 0xa0a2a3a4
<<  BMC random number             : 0xc14884760921d50fd70a0470910c6fc7
<<  BMC GUID                      : 0x20000102027f00000000000000000000
<<  Key exchange auth code [sha256]: 0x995a3dde5ddb28524a574d78c803882e9d343492bb8cba9ed7e10e892753dd2c

bmc_rand (16 bytes)
 c1 48 84 76 09 21 d5 0f d7 0a 04 70 91 0c 6f c7
>> rakp2 mac input buffer (65 bytes)
 a4 a3 a2 a0 00 e2 00 02 3e ed 21 3f ed c2 60 cf
 32 e9 6e 96 95 10 64 cc c1 48 84 76 09 21 d5 0f
 d7 0a 04 70 91 0c 6f c7 20 00 01 02 02 7f 00 00
 00 00 00 00 00 00 00 00 14 07 63 6f 6e 73 6f 6c
 65
>> rakp2 mac key (20 bytes)
 46 43 4e 42 43 31 35 32 32 30 30 34 33 35 00 00
 00 00 00 00
>> rakp2 mac as computed by the remote console (32 bytes)
 37 de bf a8 9b a7 d3 28 15 0f 48 f0 9d d5 20 b0
 d3 a3 58 b1 50 b4 e6 40 f3 72 06 f0 eb 93 3a da
> RAKP 2 HMAC is invalid
Error: Unable to establish IPMI v2 / RMCP+ session