ipmitool/ipmitool #345: Can't use ipmitool on CentOS7(VM)

no-trifling, 2022-06-13:
Can I use impitool on a virtual machine?

I carried out the following commands:
> yum install ipmitool
> modprobe ipmi_msghandler
> modprobe ipmi_devintf

But when I execute `ipmitool -I open shell`.
It still returns `Could not open device at /dev/ipmi0 or /dev/ipmi/0 or /dev/ipmidev/0: No such file or directory`.

AlexanderAmelkin, 2022-06-13:
@no-trifling if your virtual machine doesn't provide a virtual KCS device for a real KCS, then you can only use ipmitool over lan using the "-I lanplus" option. I'm sorry, I'm not aware of any VMs that virtualize KCS.

no-trifling, 2022-06-14:
Thanks, I'll try.

coklm, 2022-09-04:
Virtual machines need hypervisor implements IPMI，
These days, such a beast does exist for libvirt, called [VirtualBMC](https://github.com/openstack/virtualbmc).

closed 2022-09-04 by AlexanderAmelkin