ipmitool/ipmitool #363: Auto selecting best cipher suite is not reliable, need compare server side cipher suite's algorithms

jjqq2013, 2022-09-15:
**Describe the bug**

I was wondering why this command failed:
```
$ ~/ipmitool -H host -U admin -P adminadmin -I lanplus power status
Error in open session response message : no matching cipher suite

Error: Unable to establish IPMI v2 / RMCP+ session
```

Turns out that version 1.8.19 can automatically select best cipher suite based on what max cipher suite can server provide first,
see [lanplus.c]( https://github.com/ipmitool/ipmitool/blob/19d78782d795d0cf4ceefe655f616210c9143e62/src/plugins/lanplus/lanplus.c#L3390-L3390)
```
if (ipmi_get_channel_cipher_suites(intf, "ipmi", IPMI_LAN_CHANNEL_E,
                                   suites, &nr_suites) < 0)
{
 /* default legacy behavior - fall back to cipher suite 3 */
 return IPMI_LANPLUS_CIPHER_SUITE_3;
}
```
The ipmi_get_channel_cipher_suites returned 17 from server side.

However, **what server side means may not be compatible with client side, the authentication may fail.**

e.g.,
Server side say all my cipher suites are these:
```
[root@server ~]# ipmitool channel getciphers ipmi 1
ID   IANA    Auth Alg        Integrity Alg   Confidentiality Alg
0    N/A     none            none            none
1    N/A     hmac_sha1       none            none
2    N/A     hmac_sha1       hmac_sha1_96    none
3    N/A     hmac_sha1       hmac_sha1_96    aes_cbc_128                             ###### check here ################
6    N/A     hmac_md5        none            none
7    N/A     hmac_md5        hmac_md5_128    none
8    N/A     hmac_md5        hmac_md5_128    aes_cbc_128
11   N/A     hmac_md5        md5_128         none
12   N/A     hmac_md5        md5_128         aes_cbc_128
15   N/A     hmac_sha256     none            none
16   N/A     hmac_sha256     sha256_128      none
17   N/A     hmac_sha256     sha256_128      aes_cbc_128                                  ###### check here ################
```

Then ipmitool think oh server support suite 17, let me use it to do authentication.

Let us take a look at the client side [definition of suite 17](https://github.com/ipmitool/ipmitool/blob/19d78782d795d0cf4ceefe655f616210c9143e62/src/plugins/lanplus/lanplus.c#L259-L259
), you will know that it is not compatible with server side:

```
case IPMI_LANPLUS_CIPHER_SUITE_17:
 *auth_alg      = IPMI_AUTH_RAKP_HMAC_SHA256;
 *integrity_alg = IPMI_INTEGRITY_HMAC_SHA256_128;       ####### this does NOT match sha256_128 in id 17 of `ipmitool channel getciphers ipmi 1`
 *crypt_alg     = IPMI_CRYPT_AES_CBC_128;
```
on the contrast, the suite 3 is compatible with server side:
```
case IPMI_LANPLUS_CIPHER_SUITE_3:
 *auth_alg      = IPMI_AUTH_RAKP_HMAC_SHA1;
 *integrity_alg = IPMI_INTEGRITY_HMAC_SHA1_96;          ####### this matches hmac_sha1_96 in id 3 of `ipmitool channel getciphers ipmi 1`
 *crypt_alg     = IPMI_CRYPT_AES_CBC_128;
```

**IPMITOOL Version**

```none
$ ipmitool -V
ipmitool version 1.8.19.0.gsnapshot
```

**To Reproduce**
Steps to reproduce the behavior:
1. run ` ~/ipmitool -H host -U admin -P adminadmin -I lanplus power status`
3. Observe the abnormal reply:
    ```none
    Error in open session response message : no matching cipher suite
    Error: Unable to establish IPMI v2 / RMCP+ session
    ```

**Expected behavior**
```
Chassis Power is on
```

**Additional context**
Running `` ~/ipmitool -H host -U admin -P adminadmin -I lanplus power status -C 3` works.

AlexanderAmelkin, 2022-09-16:
If you look at IPMI 2.0 Specification, Table 22-20 "Cipher Suite IDs", you will see that there is no option for the cipher suite 17 to use 'sha256-128' instead of 'hmac-sha256-128' for integrity:

[image]

It looks like your BMC is broken and is violating the specification.

You may overcome that by specifying the desired cipher suite explicitly using the `-C 3` option.

Howerver, I'd recommend you to contact your BMC vendor for a fix. This specific problem doesn't look like an `ipmitool` bug to me.

closed 2022-09-16 by AlexanderAmelkin

AlexanderAmelkin, 2022-09-19:
@jjqq2013 also I just checked the code and can tell that what is printed as `sha256_128` in `ipmitool channel getciphers` is actually the HMAC_SHA256_128, see [lib/ipmi_strings.c#1303]:(https://github.com/ipmitool/ipmitool/blob/master/lib/ipmi_strings.c#L1297)

```c
const struct valstr ipmi_integrity_algorithms[] = {                                       
  { IPMI_INTEGRITY_NONE,         "none" },                                                
  { IPMI_INTEGRITY_HMAC_SHA1_96, "hmac_sha1_96" },                                        
  { IPMI_INTEGRITY_HMAC_MD5_128, "hmac_md5_128" },                                        
  { IPMI_INTEGRITY_MD5_128 ,     "md5_128"      },                                        
#ifdef HAVE_CRYPTO_SHA256                                                                 
  { IPMI_INTEGRITY_HMAC_SHA256_128, "sha256_128" },                                       
#endif /* HAVE_CRYPTO_SHA256 */                                                           
  { 0x00, NULL }                                                                          
};     
```

So the problem must be somewhere else, not in the non-matching protocols.
Since `ipmitool` is well able to communicate with OpenBMC that implements cipher suite 17, I still think the problem is with your BMC.