ipmitool/ipmitool #377: IPMI commands failing on "Failed to initialize the OEM info dictionary" after upgrade to 1.8.19

coklm, 2022-11-07:
**Describe the bug**
IPMI commands failing on "Failed to initialize the OEM info dictionary" after upgrade to 1.8.19

**IPMITOOL Version**
$ ipmitool -V
ipmitool version 1.8.19
```

**To Reproduce**
Steps to reproduce the behavior:
Upgrade ipmitool to 1.8.19 and re-test the same command, and observe the error:
$ ipmitool -I lanplus -H <hostname> -U <username> -P <password> chassis power status
Failed to initialize the OEM info dictionary

**Expected behavior**
The command runs successfully.

**Additional context**
I found it caused by enterprise-numbers in Makefile.am:
IANA_PEN = http://www.iana.org/assignments/enterprise-numbers
when i execute command:
curl --location --progress-bar http://www.iana.org/assignments/enterprise-numbers
it show this:
![Uploading image.png…]()

it is wrong,i think it should be: 
IANA_PEN = http://www.iana.org/assignments/enterprise-numbers.txt

alschm-lb, 2022-11-21:
same issue

AlexanderAmelkin, 2022-11-21:
Apparently, IANA have upgraded their site and changed the URL scheme.
I will update Makefile.am