ipmitool/ipmitool #383: Failed to initialize the OEM info dictionary - ipmitool 1.8.19  - MacOS Ventura 13.01

geoff-goldman, 2022-12-12:
Using Brew on MacOS Ventura 13.01 ipmitool was updated to 1.8.19 - i did not note the exact version prior to the upgrade.

Now any ipmitool command fails with the error:

Failed to initialize the OEM info dictionary

i.e.

ipmitool -I lanplus -H x.x.x.x -U abc123 -P abc123 chassis power status
Failed to initialize the OEM info dictionary

ipmitool -I lanplus -H x.x.x.x -U abc123 -P abc123 fru print 3
Failed to initialize the OEM info dictionary

vfazio, 2022-12-12:
Should be fixed by #382 

This is also likely related to https://github.com/ipmitool/ipmitool/commit/1edb0e27e44196d1ebe449aba0b9be22d376bcb6

geoff-goldman, 2022-12-12:
Thanks Vincent - removing (in my case on a Mac) /usr/local/Cellar/ipmitool/1.8.19/share/misc/enterprise-numbers did allow it to run, albeit with the 'No such file or directory' error. Will the new IANA text/html format be rolled into a new release - if that resolves #377 shouldn't it have been included in the #382 fix? Or am I not pulling the latest release via brew? Geoff

vfazio, 2022-12-12:
> Thanks Vincent - removing (in my case on a Mac) /usr/local/Cellar/ipmitool/1.8.19/share/misc/enterprise-numbers did allow it to run, albeit with the 'No such file or directory' error. Will the new [...]

AFAIK, the format didn't change, just the URL at which IANA was providing the file did. That was done _after_ 1.8.19 so that version is "broken" until the next tagged release. 

#382 resolves related issues. The fact that 1.18.19 works with a non-existent file currently is a happy accident I think, so that PR makes the behavior more uniform. It also allows for ipmitool to run if the registry file exists but is malformed.

geoff-goldman, 2022-12-12:
I guess it's always something. Thanks for the link to the text version of the PEM file! That seems to work fine with 1.8.19

geoff-goldman, 2022-12-12:
Just curious - I've been looking at earlier/older ipmitool installs and there does not seem to be a standalone enterprise-numbers files

vfazio, 2022-12-12:
> Just curious - I've been looking at earlier/older ipmitool installs and there does not seem to be a standalone enterprise-numbers files

It was generated at compile time in previous versions https://github.com/ipmitool/ipmitool/commit/bd0475ce4ad85645f893010423b265ada1514cc5#diff-931d169ad902f26c93a6af265b3bbcc571a975737f85e33ce66dea285d4d2d60L53

rhugga, 2022-12-16:
DId anyone in this thread get this working? If so what did you do? I've tried every workaround I can find to try. This broke for my entire team.

geoff-goldman, 2022-12-16:
> You can either remove the malformed file or you can download the IANA PEN
> registry file and place it in the proper path. #383 (comment)
> I don't fiddle with MacOS, but you can probably do an strace to find which
> file paths it's looking in to open the file.
> Otherwise, this should be fixed once my PR is accepted and a new version
> is released

Yes, as posted earlier I have it working - I have a MacOS brew install and
my path was /usr/local/Cellar/ipmitool/1.8.19/share/misc/enterprise-numbers
- replacing it with the text file from
https://www.iana.org/assignments/enterprise-numbers/enterprise-numbers
resolved the issue for now (thanks Vincent!)

Geoff

Geoff Goldman
***@***.***
(C) 845.797.5011

stek29, 2022-12-22:
reported to brew as Homebrew/homebrew-core#118705