ipmitool/ipmitool #385: Wrong DIMMs given in SEL with Quanta QCT HW

JPlanche, 2022-12-16:
Hello,

Sorry for the delay before posting this bug.

Since #5 has been merged, `SEL elist` is giving wrong informations when we have CE or UE errors on some Quanta QCT hardware.

```none
$ ipmitool -V
ipmitool version 1.8.19
```
(MacOS  + Homebrew)

**Hardware Used**
```none
 Board Mfg             : Quanta Cloud Technology Inc.
 Board Product         : S5HF MB (QCT)
 Product Manufacturer  : Quanta Cloud Technology Inc.
 Product Name          : QuantaPlex T22HF-1U
```

**ipmitool sel elist**
```
  3a | 12/14/22 | 23:09:44 CET | Memory #0x88 | Correctable ECC (CPU0_G1) | Asserted
```

**ipmiutil/raw**
```
003a 12/14/22 23:09:44 MIN EFI  Memory #88  Correctable ECC, DIMM(49) 6f [a0 02 31]
```

**BMC web interface displays**
```
58
Dec/14/2022
22:09:44
[Information]
[Memory Error AMD]
[Memory]
Correctable ECC(CPU0_E1) - Asserted
```

Usage has proven the BMC web interface always speaks truth.
I have also a technical PDF from QCT that matches with what BMC WUI displays.

However, on another hardware, DIMM labels are not displayed:
```
 Product Manufacturer  : Quanta Cloud Technology Inc.
 Product Name          : QuantaMicro X10E-9N
 
    e | 06/10/22 | 19:31:43 CEST | Memory #0x87 | Correctable ECC () | Asserted
```
Which is fine, I prefer no information than false information.

Let me know if I can help with the some faculties I have!

doublechiang, 2023-02-19:
Not sure if I understand this issue correctly.
So there is no DIMM info decoded for platform X10E-9N, and you are fine with that, correct?
Then why the tile of this issue mentioned ti's wrong DIMM Infomation?

JPlanche, 2023-02-19:
> Not sure if I understand this issue correctly. So there is no DIMM info decoded for platform X10E-9N, and you are fine with that, correct? Then why the tile of this issue mentioned ti's wrong DIMM Infomation?

I have wrong DIMM information with T22HF-1U / S5HF since https://github.com/ipmitool/ipmitool/pull/5, hence this report.

doublechiang, 2023-02-20:
Let me confirm, so on S5HF, either no DIMM information reported, or reported correctly, works for you, right?  
In this case, would you like to contact FAE or the sales at Quanta, and let them know this patch https://github.com/ipmitool/ipmitool/pull/5 doesn't apply for S5HF and it results in faulty information?