ipmitool/ipmitool #386: 1.8.19 broken on macos

rhugga, 2022-12-16:
Updated another brew package, it also updated ipmitool without warning or desire, resulting in a broken ipmitool 1 hour before this was needed for a VP-level demo. Good times.

Failed to initialize the OEM info dictionary

Spent 2 hrs trying various hacks and work-arounds to which none worked.

IS THERE A WAY TO DOWNREV THIS? "brew search ipmitool" shows no other versions. All I get now is 1.8.19. It's broken for my entire team.

So far this has affected the current and two most recent macos versions. I'm personally on 12.5.1. (monterey)

On a side note, does anyone know of another tool thats does this or another way to install ipmitool outside of brew that might have proper version history and ability to downrev?

nsoggia, 2022-12-19:
Makefile.am downloads the database from the wrong url.

http://www.iana.org/assignments/enterprise-numbers
should be:
https://www.iana.org/assignments/enterprise-numbers.txt

vfazio, 2022-12-19:
This is a duplicate of #383

stek29, 2022-12-22:
see Homebrew/homebrew-core#118705