ipmitool/ipmitool #388: Unsupported LAN Parameter lookup command returns an error

dvanallen, 2022-12-21:
When using `ipmitool lan print`, an error (exit status 1) is returned for an Unsupported Parameter in versions which include the change https://github.com/ipmitool/ipmitool/commit/351dad24a26f56580ba6028a267c4924b7faa4cb.

**IPMITOOL Version**

```none
ipmitool -V
ipmitool version 1.8.19.4.g63d59a5
```

**To Reproduce**
Steps to reproduce the behavior:
1. Using Dell R740 server with Linux 5.15
2. Run `ipmitool -vvvvv lan print`
3. Observe the abnormal log message and exit code 1:
    ```none
    Get LAN Parameter 'Bad Password Threshold' command failed: Unsupported parameter
    ```

**Expected behavior**
Function `get_lan_param_select` should not return NULL, as described in the function comment at https://github.com/ipmitool/ipmitool/blob/63d59a50a6809e415cd7d8915533f236f28865f3/lib/ipmi_lanp.c#L179-L181. The `ipmitool lan print` command should return exit code 0.

osenft, 2022-12-21:
I expect this to be an issue on all current Dell machines with iDRAC BMC, since AFAIK these do not implement the "bad password threshold" parameter.

rustydb, 2023-02-02:
We are also seeing this on iLO.

```bash
Set in Progress         : Set Complete
Auth Type Support       : NONE MD5 PASSWORD
Auth Type Enable        : Callback : MD5 PASSWORD
                        : User     : MD5 PASSWORD
                        : Operator : MD5 PASSWORD
                        : Admin    : MD5 PASSWORD
                        : OEM      :
IP Address Source       : Static Address
IP Address              : 10.100.254.41
Subnet Mask             : 255.255.255.0
MAC Address             : 94:40:c9:37:77:ce
SNMP Community String   :
```
with `-v`:
```bash
Loading IANA PEN Registry...
Using best available cipher suite 3

Running Get VSO Capabilities my_addr 0x20, transit 0, target 0x20
Invalid completion code received: Invalid command
Discovered IPMB address 0x0
Set in Progress         : Set Complete
Auth Type Support       : NONE MD5 PASSWORD
Auth Type Enable        : Callback : MD5 PASSWORD
                        : User     : MD5 PASSWORD
                        : Operator : MD5 PASSWORD
                        : Admin    : MD5 PASSWORD
                        : OEM      :
IP Address Source       : Static Address
IP Address              : 10.100.254.41
Subnet Mask             : 255.255.255.0
MAC Address             : 94:40:c9:37:77:ce
SNMP Community String   :
Get LAN Parameter 'IP Header' command failed: Unsupported parameter
```

```
rpm -q ipmitool
ipmitool-1.8.18.238.gb7adc1d-150400.1.14.x86_64
```

We've worked around it by rolling back to 7ccea283dd62a05a320c1921e3d8d71a87772637, which is the commit hash SLES15SP3 used for its `ipmitool` package. https://build.opensuse.org/package/show/SUSE:SLE-15-SP3:Update/ipmitool

The invalid parameter still occurs but it doesn't exit early.

rustydb, 2023-02-17:
I tested the change in #389 against the current HEAD of ipmitool/ipmitool and it resolved the issue for us with regards to our iLO BMCs. Errors are thrown when `-v` (verbose) is used, but the command does not return early.

```bash
redbull-pit:~ # rpm -q ipmitool
ipmitool-1.8.19.ga61349b-0.x86_64
redbull-pit:~ # ipmitool -I lanplus -U root -P initial0 -H surtur-ncn-m001-mgmt.hpc.amslabs.hpecorp.net lan print 1
Set in Progress         : Set Complete
Auth Type Support       : NONE MD5 PASSWORD
Auth Type Enable        : Callback : MD5 PASSWORD
                        : User     : MD5 PASSWORD
                        : Operator : MD5 PASSWORD
                        : Admin    : MD5 PASSWORD
                        : OEM      :
IP Address Source       : Static Address
IP Address              : 10.100.254.79
Subnet Mask             : 255.255.255.0
MAC Address             : 94:40:c9:37:87:32
SNMP Community String   :
BMC ARP Control         : ARP Responses Enabled, Gratuitous ARP Disabled
Default Gateway IP      : 10.100.254.1
802.1q VLAN ID          : Disabled
802.1q VLAN Priority    : 0
RMCP+ Cipher Suites     : 0,1,2,3
Cipher Suite Priv Max   : XuuaXXXXXXXXXXX
                        :     X=Cipher Suite Unused
                        :     c=CALLBACK
                        :     u=USER
                        :     o=OPERATOR
                        :     a=ADMIN
                        :     O=OEM
Bad Password Threshold  : Not Available
redbull-pit:~ # ipmitool -v -I lanplus -U root -P initial0 -H surtur-ncn-m001-mgmt.hpc.amslabs.hpecorp.net lan print 1
Loading IANA PEN Registry...
Using best available cipher suite 3

Running Get VSO Capabilities my_addr 0x20, transit 0, target 0x20
Invalid completion code received: Invalid command
Discovered IPMB address 0x0
Set in Progress         : Set Complete
Auth Type Support       : NONE MD5 PASSWORD
Auth Type Enable        : Callback : MD5 PASSWORD
                        : User     : MD5 PASSWORD
                        : Operator : MD5 PASSWORD
                        : Admin    : MD5 PASSWORD
                        : OEM      :
IP Address Source       : Static Address
IP Address              : 10.100.254.79
Subnet Mask             : 255.255.255.0
MAC Address             : 94:40:c9:37:87:32
SNMP Community String   :
Get LAN Parameter 'IP Header' command unsupported: Unsupported parameter
BMC ARP Control         : ARP Responses Enabled, Gratuitous ARP Disabled
Get LAN Parameter 'Gratituous ARP Intrvl' command unsupported: Unsupported parameter
Default Gateway IP      : 10.100.254.1
Get LAN Parameter 'Default Gateway MAC' command unsupported: Unsupported parameter
Get LAN Parameter 'Backup Gateway IP' command unsupported: Unsupported parameter
Get LAN Parameter 'Backup Gateway MAC' command unsupported: Unsupported parameter
802.1q VLAN ID          : Disabled
802.1q VLAN Priority    : 0
RMCP+ Cipher Suites     : 0,1,2,3
Cipher Suite Priv Max   : XuuaXXXXXXXXXXX
                        :     X=Cipher Suite Unused
                        :     c=CALLBACK
                        :     u=USER
                        :     o=OPERATOR
                        :     a=ADMIN
                        :     O=OEM
Get LAN Parameter 'Bad Password Threshold' command unsupported: Unsupported parameter
Bad Password Threshold  : Not Available
```

rustydb, 2023-02-17:
I tested the change in #389 against the current HEAD of ipmitool/ipmitool and it resolved the issue for us with regards to our iLO BMCs. Errors are thrown when `-v` (verbose) is used, but the command does not return early.