ipmitool/ipmitool #397: windows server2016 Err:no hostname specified

zhangbob902, 2023-02-18:
**Describe the bug**

Windows server 2016 uses ipmitool to access bmc in-band, and an error no hostname specified occurs.
configure result:
./configure  --enable-intf-lanplus=yes --enable-intf-imb=yes IANADIR=.local/usr/share/misc
ipmitool 1.8.19

Interfaces (default=lan)
ipmitool 1.8.19

Interfaces (default=lan)
  lan     : yes
  lanplus : yes
  open    : no
  free    : no
  imb     : yes
  bmc     : no
  dbus    : no
  usb     : no
  lipmi   : no
  serial  : yes
  dummy   : no

Extra tools
  ipmievd   : yes
  ipmishell : yes

Extra tools
  ipmievd   : yes
  ipmishell : yes

**IPMITOOL Version**

version:1.8.19

```none
$ ipmitool -V
PS C:\Users\Administrator\Desktop\ipmitool-master\ipmitool-master\ipmitool-> .\ipmitool.exe -V
ipmitool version 1.8.19
PS C:\Users\Administrator\Desktop\ipmitool-master\ipmitool-master\ipmitool-> .\ipmitool.exe raw 6 1
No hostname specified!
```
windows drivers：
PS C:\Users\Administrator> driverquery.exe

模块名       显示名称               驱动程序类型  链接日期
============ ====================== ============= ===================
1394ohci     1394 OHCI Compliant Ho Kernel        2016/7/16 10:21:36
3ware        3ware                  Kernel        2015/5/19 6:28:03
ACPI         Microsoft ACPI Driver  Kernel        2016/7/16 10:10:47
AcpiDev      ACPI 设备驱动程序      Kernel        2016/7/16 10:29:10
acpiex       Microsoft ACPIEx Drive Kernel        2016/7/16 10:28:23
acpipagr     ACPI 处理器聚合器驱动  Kernel        2016/7/16 10:29:00
AcpiPmi      ACPI 电源表驱动程序    Kernel        2016/7/16 10:19:44
acpitime     ACPI 唤醒警报驱动程序  Kernel        2016/7/16 10:29:20
ADP80XX      ADP80XX                Kernel        2015/4/10 4:49:48
AFD          Ancillary Function Dri Kernel        2016/10/15 11:53:45
ahcache      Application Compatibil Kernel        2016/10/15 11:31:36
AmdK8        AMD K8 Processor Drive Kernel        2016/7/16 10:10:42
AmdPPM       AMD Processor Driver   Kernel        2016/7/16 10:10:41
amdsata      amdsata                Kernel        2015/5/14 20:14:52
amdsbs       amdsbs                 Kernel        2012/12/12 5:21:44
amdxata      amdxata                Kernel        2015/5/1 8:55:35
AppID        AppID 驱动程序         Kernel        2016/7/16 10:27:05
applockerflt Smartlocker 筛选器驱动 Kernel        2016/7/16 10:27:27
AppvStrm     AppvStrm               File System   2016/9/16 0:15:25
AppvVemgr    AppvVemgr              File System   2016/7/16 10:10:56
AppvVfs      AppvVfs                File System   2016/7/16 10:10:53
arcsas       Adaptec SAS/SATA-II RA Kernel        2015/4/10 3:12:07
AsyncMac     RAS 异步媒体驱动程序   Kernel        2016/7/16 10:29:00
[... 252 lines cut ...]
wcifs        Windows Container Isol File System   2016/9/16 0:42:03
wcnfs        Windows Container Name File System   2016/7/16 10:28:27
WdBoot       Windows Defender Boot  Kernel        2016/7/16 10:28:51
Wdf01000     Kernel Mode Driver Fra Kernel        2016/7/16 10:13:12
WdFilter     Windows Defender Mini- File System   2016/7/16 10:25:21
WdNisDrv     Windows Defender Netwo Kernel        2016/7/16 10:25:56
WFPLWFS      Microsoft Windows 筛选 Kernel        2016/7/16 10:25:57
WIMMount     WIMMount               File System   2016/7/16 10:24:25
WindowsTrust Windows Trusted Execut Kernel        2016/7/16 10:27:05
WindowsTrust Microsoft Windows 可信 Kernel        2016/7/16 10:27:05
WinMad       WinMad 服务            Kernel        2016/4/10 21:46:08
WinNat       Windows NAT Driver     Kernel        2016/7/16 10:26:53
WINUSB       WinUsb Driver          Kernel        2016/7/16 10:28:32
WinVerbs     WinVerbs 服务          Kernel        2016/4/10 21:46:10
WmiAcpi      Microsoft Windows Mana Kernel        2016/7/16 10:18:19
Wof          Windows Overlay File S File System   2016/8/6 11:45:24
WpdUpFltr    WPD Upper Class Filter Kernel        2016/7/16 10:27:41
ws2ifsl      Winsock IFS Driver     Kernel        2016/7/16 10:27:26
WudfPf       User Mode Driver Frame Kernel        2016/7/16 10:26:10
WUDFRd       WUDFRd                 Kernel        2016/7/16 10:25:31
WUDFWpdFs    WUDFWpdFs              Kernel        2016/7/16 10:25:31
xboxgip      Xbox Game Input Protoc Kernel        2016/11/11 17:26:51
xinputhid    XINPUT HID Filter Driv Kernel        2016/8/20 13:20:50

thanks.

JPlanche, 2023-02-18:
`Interfaces (default=lan)`

I believe that ipmitool wants here to use lan by default, so as you do not give any IP nor hostname, it fails.
You should be using some driver to communicate with the BMC.
Maybe ask the manufacturer of your motherboard, or see their documentation.
You may have (maybe) some luck with https://ipmiutil.sourceforge.net/

I do not think it is a bug.

zhangbob902, 2023-02-20:
> `Interfaces (default=lan)`
>
> I believe that ipmitool wants here to use lan by default, so as you do not give any IP nor hostname, it fails. You should be using some driver to communicate with the [...]

The parameter lan/lanplus is required for ipmi to access through the network, and the in-band access to bmc is  KCS. I am not sure what is the difference between windows and linux, and linux is normal，